OctoPrint/OctoPrint #3911: Broken URL

sixtyfive, 2020-12-27:
> There are [hints in the wiki](https://github.com/foosel/OctoPrint/wiki/Setup-on-a-Raspberry-Pi-running-Raspbian#automatic-start-up) on how to get MJPG-Streamer started up on boot […]

_Originally posted by @foosel in https://github.com/OctoPrint/OctoPrint/issues/335#issuecomment-31359952_

Given that Google finds the above Issue, but doesn't find the current URL to those hints, what _is_ the current URL?

jneilliii, 2020-12-27:
The majority of the wiki stuff has been moved to the knowledgebase. You're looking for what's part of this page I believe. It's listed in the optional webcam part.

https://community.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian-or-raspberry-pi-os/2337?u=jneilliii

closed 2020-12-27 by jneilliii

sixtyfive, 2020-12-27:
Thank you @jneilliii. Hopefully, Google will pick this up, too.