OctoPrint/OctoPrint #3914: Communication timeout during an active resend / Printer keeps requesting line 13 again

tarsonis123, 2020-12-29:
### What were you doing?
Print through OctoPrint with Cura

Ideally provide exact steps to follow in order to reproduce your problem:
Hit "Print with OctoPrint in Cura"

#### What did you expect to happen?
Well it should print and not stall. This issue is kind of new and I didn't change the firmware.

#### What happened instead?
The print stalled with the code saying:
`Communication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.`

`Send: N12 G1 X1 Y-11 Z2 F12000*27
Recv: Error:Line Number is not Last Line Number+1, Last Line: 12
Recv: Resend: 13
Recv: ok
Send: N12 G1 X1 Y-11 Z2 F12000*27
Recv: Error:Line Number is not Last Line Number+1, Last Line: 12
Recv: Resend: 13
Printer keeps requesting line 13 again and again, communication stuck
Changing monitoring state from "Cancelling" to "Error: Printer keeps requesting line 13 again and again, communication stuck"`

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint
1.5.2

#### Operating System running OctoPrint
Win 10 on Workstation via USB

#### Printer model & used firmware incl. version
Ender 3 with Marlin 2.0.7.2 stable

#### Browser and version of browser, operating system running browser
Firefox 84.0.1

#### Link to octoprint.log
https://www.dropbox.com/s/7per019e297uzlk/octoprint.log?dl=0

#### Link to contents of terminal tab or serial.log
`Communication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.`

`Send: N12 G1 X1 Y-11 Z2 F12000*27
Recv: Error:Line Number is not Last Line Number+1, Last Line: 12
Recv: Resend: 13
Recv: ok
Send: N12 G1 X1 Y-11 Z2 F12000*27
Recv: Error:Line Number is not Last Line Number+1, Last Line: 12
Recv: Resend: 13
Printer keeps requesting line 13 again and again, communication stuck
Changing monitoring state from "Cancelling" to "Error: Printer keeps requesting line 13 again and again, communication stuck"`

#### Link to contents of Javascript console in the browser
?

#### Screenshot(s)/video(s) showing the problem:

Here is the start gcode for my printer. Seems related but this is all more or less cryptic to me what the terminal tries to say

```
G90 ;absolute positioning
M82 ;set extruder to absolute mode

; Marlin 2.0 features
;M900 K0.50 ; Set K Value
;M205 J0.09 ; Set Junction Deviation

M117 Lets Go. ; Print a message on the LCD.

;Manual/Mesh Bed Leveling version;
M140 S{material_bed_temperature} ;set bed temperature
M190 S{material_bed_temperature} ;wait for bed temperature

G28 			        ;home all axis
M420 L0 S1 Z1		        ;use mesh level

G1 X1 Y-11 Z2 F12000    ;Move X on the bed

M104 S{material_standby_temperature} ;set hotend to standby temperature
M109 S{material_print_temperature} ;wait for hotend temperature
M400 ;Finish moves before continuing
M300 S220 P100
M117 Sensors Online.

; Purge sequence
M400
M300 S440 P200
M117 Weapons Online.

G91
G1 F150 E15 ; Extrude 15mm in the air
G90
G1 Y10 F3000  ;Move Y to knock the filament off
M400
G1 Z0.2 F500 ; Drop the Z Back
G91
G1 Y100 F750 E3 ; Extrude to the build plate
G90

M400
M300 S440 P200

G1 Y110 F4000 ; Quick Wipe

M400
;M300 S660 P100	;Play tone

G1 Z2.5 F1500 ; Lift the Z
G92 E0

M420 L0 S1 Z1
;M300 S440 P50	;Play tone
M117 Yeah Ready
```

cp2004, 2020-12-29:
You appear to have significant communication issues with your printer. It seems to ok lines, then ask for them to be resent, and then still get stuck again. It also seems to require resends after it levels the bed regularly. It also sometimes persistently asks for the *same* line over and over again.

This doesn't seem to be a problem with OctoPrint - you probably need a better USB cable or firmware. You mention Marlin 2.0.7.2 which is better than Creality FW at having less issues. Although there does seem to be confusion between the two over which line they are talking about, it is not obvious where the root of the problem is.

Please **enable and upload the serial.log** (note you will have to reproduce the problem first) so we can see the full picture of what is going back and forth between the printer. If it still looks like general communication issues, we might suggest opening a topic on the [community forums](https://community.octoprint.org) or heading to the [discord server](https://discord.octoprint.org), since it is not an issue with OctoPrint but your general setup.

tarsonis123, 2020-12-29:
Thanks for looking into it. It did work the last week without flaws. Since yesterday it started with this connection errors and since there were updates. I don't remember if it were Octoprints plugins or Octoprint itself I thought this could be related. Those errors from the terminal are cryptic to me so I have no clue what it want from me.
I gonna try the serial.log thing. Reproducing it is easy. When I raise the connection time out it happens instantly and never prints after the "heat extruder" command

tarsonis123, 2020-12-30:
Ok. So today the issue occurred again after I did some probing. I send a print through Cura which immediately lead to the communication error. I then pressed "pause" in Cura and immediately resume after that. Got the error message again once but the print then started.
I got the serial.log for this. I cant see anything pointing to the source of the problem.
[serial.log](https://github.com/OctoPrint/OctoPrint/files/5754301/serial.log)

cp2004, 2021-01-04:
Apologies, missed your comment about the serial log, marked as approved for now so the bot doesn't close it.
```
2020-12-30 14:09:10,144 - Send: N5 M140 S85*93
2020-12-30 14:09:10,146 - Recv: ok
2020-12-30 14:09:10,147 - Send: N6 M140 S85.0*64
2020-12-30 14:09:10,148 - Recv: ok
2020-12-30 14:09:12,320 - Recv:  T:29.15 /0.00 B:69.75 /85.00 @:0 B@:127
2020-12-30 14:09:15,317 - Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
2020-12-30 14:09:15,320 - Recv:  T:29.29 /0.00 B:69.99 /85.00 @:0 B@:127
```
I wonder if the problem is the `M140 S85.0` section. Not entirely sure why that would be repeated, the `M140 S85` is fine, then it begins to go wrong from there.

Still don't think that this is a bug in OctoPrint, but something either wrong with your slicer setup (maybe start scripts?) or printer firmware, I personally have not had many issues with Marlin 2.0.7.2 but it is *definitely* not without it's bugs.

As for the default connection settings, look at the [configuration docs](https://docs.octoprint.org/en/master/configuration/config_yaml.html#serial) for the settings.

tarsonis123, 2021-01-04:
Thanks for the follow up. You are right that it seems to be be related to the start code. The last 4 times I got the issue the connection error message by Octoprint started after these stand-by commands:

M190 S{material_bed_temperature} ;wait for bed temperature
M109 S{material_print_temperature} ;wait for hotend temperature

Not quiet sure how to deal with it as its seems random or at least related to power cycles. Since I switched to UBL the ABL function does fortunately and finally work reliable and consistent. No crashing into the bed. So I didn't need to do a lot of power cycles since then and this issue happens less.

foosel, 2021-03-04:
Sorry for this not yet having seen a comment by me, I was hoping I could provide some "oh, I know what's up" but so far I'm drawing a blank here...

foosel, 2021-03-09:
So, the only things I can tell you right now is that it looks indeed related to the auto temp report interval.

In [your first `serial.log`](https://github.com/OctoPrint/OctoPrint/issues/3914#issuecomment-752585231), it was set to 3s right before starting the print:

```
2020-12-30 14:04:36,284 - Send: M155 S3
2020-12-30 14:04:36,285 - Recv: ok
[...]
2020-12-30 14:09:10,116 - Changing monitoring state from "Operational" to "Starting"
2020-12-30 14:09:10,125 - Send: N0 M110 N0*125
2020-12-30 14:09:10,128 - Recv: ok
2020-12-30 14:09:10,128 - Changing monitoring state from "Starting" to "Printing"
2020-12-30 14:09:10,130 - Send: N1 M82*24
2020-12-30 14:09:10,133 - Recv: ok
2020-12-30 14:09:10,135 - Send: N2 G90*18
2020-12-30 14:09:10,137 - Recv: ok
2020-12-30 14:09:10,137 - Send: N3 M82*26
2020-12-30 14:09:10,141 - Recv: ok
2020-12-30 14:09:10,142 - Send: N4 M117 Lets Go.*8
2020-12-30 14:09:10,144 - Recv: ok
2020-12-30 14:09:10,144 - Send: N5 M140 S85*93
2020-12-30 14:09:10,146 - Recv: ok
2020-12-30 14:09:10,147 - Send: N6 M140 S85.0*64
2020-12-30 14:09:10,148 - Recv: ok
2020-12-30 14:09:12,320 - Recv:  T:29.15 /0.00 B:69.75 /85.00 @:0 B@:127
2020-12-30 14:09:15,317 - Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
2020-12-30 14:09:15,320 - Recv:  T:29.29 /0.00 B:69.99 /85.00 @:0 B@:127
2020-12-30 14:09:18,317 - Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
```

With a default communication timeout of 3s if `busy` protocol is enabled that obviously will cause issues, way too close together, which is why the default for the auto temp report interval is **2**s (set via `M155` and also to be seen in your log right on connect, before it then gets constantly toggled between a whopping 30s and the aforementioned 3s before print begin), and the default for the busy report interval (set via `M113`) is `[communication timeout w/ busy] - 1s`, so `3s - 1s = 2s` by default.

Please check what you have configured under Settings > Serial Connection > Intervals & timeouts and ensure those values make sense - according to the log they didn't seem to. As a rule of thumb, the auto temp interval must be smaller than both the communication timeout with and without `busy` support, if autotemp is enabled. Default values are 2s autotemp, 30s w/o `busy`, 3s w/ `busy`. 

In [your latest `serial.log`](https://github.com/OctoPrint/OctoPrint/issues/3914#issuecomment-754111141) however I frankly don't see any issues at all.

tarsonis123, 2021-03-09:
Thanks for investigating into my issue. Those timeout values were from a journey finding the right setting that is not leading to this issue. The default values were the starting point. Wouldn't I have had those problems I wouldn't have change those values in the first place of course.
In the end I couldnt resolve this issue and finally my Pi 3b+ arrived which is running most of the time flawless since then. It's connected to the SKR E3 2 via GPIO and it's running Klipper since some weeks. So no problematic USB connection anymore. 
Since then I never had this specific connection problem. Well Win10 and the USB connection is out of the equation so who knows what was causing this constant problem. Anyway thanks for trying to help here.

foosel, 2021-03-09:
Ok, closing this then. Glad to hear things are running smoothly now.