OctoPrint/OctoPrint #3915: mandatory access control should not be mandatory

draeath, 2020-12-31:
I run octoprint on a closed network, as do many others. Why is this required, now? This is an annoyance, and _now we have to patch it out ourselves if we don't want it._ This sort of thing should be left up to the users, and not imposed upon them.

I'd suggest a configuration (not exposed in the interface, if you like) to disable this. It might do to include a filter such that only local subnet hosts are allowed access when this is enabled, to guard against it somehow being exposed.

draeath, 2020-12-31:
A note: making this mandatory means that, if we had set up more elaborate or flexible authentication in front of octoprint (via an apache proxy, for instance) we can no longer do so (or it's redundant). For instance, had I wanted to use LDAP authentication, that's now impossible.

closed 2020-12-31 by draeath

foosel, 2020-12-31:
There's autologin to bypass authentication for trusted ip networks, and there's more elaborate login header evaluation as well. All of that is [documented](https://docs.octoprint.org/en/master/configuration/config_yaml.html#access-control) and has been for years as it's not even new functionality. None of your scenarios require patching, they only require configuration and that's all readily linked in the release notes and announcement as well, as is the reasoning why this was necessary. I suggest you read those, they should make everything clear.

For reference:
- https://github.com/OctoPrint/OctoPrint/releases/tag/1.5.0
- https://community.octoprint.org/t/how-to-set-up-octoprint-to-autologin-a-single-user-when-connecting-from-the-internal-network/26235

cp2004, 2020-12-31:
I think they might have found the autologin config plugin, since they've commented there on the request about auto-detecting the subnet etc. To which, I will go and answer the question now.

draeath, 2021-01-06:
Yep, that's the case. Thanks, and sorry for the bother.