OctoPrint/OctoPrint #3920: No UI after login, iPad 2 -1.5.2

ITASJHall, 2021-01-02:
#### What were you doing?

1. Go to the ip of the Octoprint on iPad via Safari, Chrome, or firefox
2. Click on enter credentials and login
3. Scroll down to '....'
4. See error

#### What did you expect to happen?
 To load the UI of Octoprint
#### What happened instead?
 After login is displays a blank white screen
#### Did the same happen when running OctoPrint in safe mode?
Yes, same on all browsers as well 

#### Version of OctoPrint
1.5.2

#### Operating System running OctoPrint
OctoPi

#### Printer model & used firmware incl. version
Anet A8 marlin

#### Browser and version of browser, operating system running browser
Safari, Chrome, Firfox on iPad 2 running iOS 9.3.5

#### Link to octoprint.log
https://pastebin.com/Yu55sCat

#### Link to contents of terminal tab or serial.log
N/A

#### Link to contents of Javascript console in the browser
Not sure on how to do that on iPad but this might help
https://imgur.com/a/Ze5s1aW

#### Screenshot(s)/video(s) showing the problem:
https://puu.sh/H30U8/9c50159daf.png

I have read the FAQ.

jneilliii, 2021-01-02:
Your errors indicate the inability to load assets into the browser. Does the browser or ipad have any additional security software installed or settings that prevent it from loading javascript or storing cookies?

jneilliii, 2021-01-02:
@cp2004 I can't seem to assign issues to projects, but this one looks complete to me.

ITASJHall, 2021-01-02:
As far as i know JavaScript is enabled. I'll get some more logs. I've seen a couple others have the same issue. 
https://www.reddit.com/r/octoprint/comments/k4o77b/octoprintversion_150_ipads_cannot_load_thru/

ITASJHall, 2021-01-02:
I finally found an app to get console log data.
This is what my OctoPrint is showing after login

http://192.168.1.50/
http://192.168.1.50/login/?redirect=%2F%3F&permissions=STATUS%2CSETTINGS_READ
http://192.168.1.50/?
SyntaxError: Strict mode does not allow function declarations in a lexically nested statement.
packed_libs.js:1249

cp2004, 2021-01-02:
> Strict mode does not allow function declarations in a lexically nested statement.

Is strict mode a setting in the browser?

iPad 2, that puts us at iOS 9? Maybe something in the JS dependencies dropped compatibility with that. I didn't notice, but will check.

---
> @cp2004 I can't seem to assign issues to projects, but this one looks complete to me.

Yeah, we don't have project permissions on octoprint/octoprint-triage to access project boards.

jneilliii, 2021-01-02:
Thanks. Yeah a lot of the errors I saw related to this error when searching seemed to be older iOS and react related surprisingly.

tech360, 2021-01-02:
I'm having the same issue.  I just pulled an old iPad 2 with ios 9.3.5 loaded on it.  I could not get any browser to work past the login screen for octoprint.  Even touch ui would not work.

cp2004, 2021-01-03:
I've got an old iPad mini in the drawer here, will test and see if something useful happens.

cp2004, 2021-01-04:
Update: we got into the iPad, so it's connected & I have access to the devtools. Also in the meantime got a BrowserStack license, but the physical device is easier.

The problem appears to be with the less.min.js dependency in the frontend. However, due to OctoPrint's caching, minifying and bundling the syntax error takes out the *entire* javascript suite, and it never recovers.

@ITASJHall @tech360 please try the below workaround, if you can. You need to set the webassets to *not* minify or bundle.
In `config.yaml`, add the following section:
```
devel:
    webassets:
        bundle: false
        minify: false
```
**Please note that this will slow down your load times on all devices**.

Browser's error console when unminified:
```
VMundefined less.min.js:11 SyntaxError: Strict mode does not allow function declarations in a lexically nested statement.
(anonymous) @ VMundefined less.min.js:11
VMundefined less.min.js:31 SyntaxError: Unexpected identifier 'param'. Expected either 'in' or 'of' in enumeration syntax.
(anonymous) @ eeprom_marlin.js:31
VMundefined less.min.js:65 SyntaxError: Unexpected identifier 'cap'. Expected either 'in' or 'of' in enumeration syntax.
(anonymous) @ virtual_printerconfig.js:65
VMundefined less.min.js:14 SyntaxError: Unexpected token '='. Expected a ')' or a ',' after a parameter declaration.
```
Interestingly two of the above errors are from my own plugins, so I tried this in safe mode.
Browser console in safe mode:
```
VMundefined less.min.js:11 SyntaxError: Strict mode does not allow function declarations in a lexically nested statement.(anonymous) @ VMundefined less.min.js:11
```
Other than that, the full OctoPrint UI appears to work, and there are no other errors.

---
I went a little further, and replaced the minified less file with the full version, to see *exactly* where the error lies:
```
VMundefined less.min.js:8946 SyntaxError: Strict mode does not allow function declarations in a lexically nested statement.
```

ITASJHall, 2021-01-05:
> devel:
>     webassets:
>         bundle: false
>         minify: false
> ```
Yeah, so added the code to config.yaml and restarted the service. Now I'm getting the "OctoPrint server is currently not running"
after removing it it still can up with "OctoPrint server is currently not running"
cp
It doesn't look like the python service is running and restarting the pi didn't help either.
The log file didn't seem to be populating either

jneilliii, 2021-01-05:
Yeah, config.yaml is extremely sensitive to space indentation. You have to be very careful. For our knowledge, did setting those values allow you to load the interface again now on your iPad?

ITASJHall, 2021-01-05:
Yes, it is loading on the iPad now. TouchUI doesn't look that great but it does load.

unicornhero22, 2021-01-10:
How can i modify the config.yaml, or i cant find it

foosel, 2021-01-11:
I'm not so sure it's less.min.js. It's more the problem that for some reason in this browser strict mode is enabled for something it shouldn't be enabled for. I can reproduce on Browserstack, and I can trigger it even earlier with a plugin if not running in safe mode. What I don't understand is what is settings the JS parser to strict mode in the first place here.

foosel, 2021-02-09:
This should hopefully no longer be happening. The issue is that the ancient safari in question a) can't cope at ALL with anything but ES5 and b) also seems to have a somewhat restrictive interpretation of `strict` mode that I can't find documented anywhere.

In any case, I've made sure to add an eslint config that should raise alarm when stuff is committed that is not ES5 compatible, updated `less.js` and removed a safari `strict` mode violation, and just in case also added an ES6 polyfill that should do its best to keep plugins working that don't check for ES5 compatibility. Alas, that won't magically add support for stuff like `let`, `const` or arrow functions. I briefly tried to add a babeljs transpiler filter to the webassets bundler using dukpy, but that didn't work at all (server just... hung, couldn't figure out why).

commit 473d72f5 (2021-02-08): 🗑 Better backwards compatibility with ancient browsers

Added an ES6 polyfill, fixed some minor issues in less.js, put less.js
into lib bundle again.

Further steps should be taken to ensure the core JS files stay ES5
compliant with regards to the use of arrow functions, let, const and so
on, at least the versions bundled into the webassets.

Closes #3920
closed 2021-04-27 by foosel