OctoPrint/OctoPrint #3922: Z-axis does not go back to intended height after resuming on pause. Prints in mid-air.

SysGh-st, 2021-01-02:
#### What were you doing?
Printing as usual. Had to pause the print for a few seconds.
When paused, the Z-axis raises a few millimeters and then the head goes to origin.
When resuming, the Z-axis does not go back down, but instead start printing "in the air" those millimeters above.

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

#### What did you expect to happen?
Expected the Z-Axis to go back down to the layer where it got paused at.

#### What happened instead?
Z-axis never got back down. Instead it resumed with a gap between the print and the head.

#### Did the same happen when running OctoPrint in safe mode?
Yes.

#### Version of OctoPrint
    OctoPrint 1.5.2
    Python 3.7.3
    OctoPi 0.18.0

#### Operating System running OctoPrint
Linux on Raspberry pi. using OctoPi 0.18.0

#### Printer model & used firmware incl. version
Creality Ender 3, Latest Marlin Firmware (bugfix-2.0.x as of 2020-12-22)

#### Browser and version of browser, operating system running browser
Firefox 84.0 (64-bit)
Archlinux

#### Link to octoprint.log
https://gist.github.com/SysGh-st/405652319af1b30899641baf76297439

#### Link to contents of terminal tab or serial.log
"2021-01-02 05:11:52,873 - serial.log is currently not enabled, you can enable it via Settings > Serial Connection > Log communication to serial.log"
Got nothing but temperatures logs in the terminal tab

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

cp2004, 2021-01-02:
This is an issue with your gcode scripts, you must have set one that moves the head up a few mm on pause. Read the [documentation on gcode scripts](https://docs.octoprint.org/en/master/features/gcode_scripts.html#gcode-scripts) to find out how you can use the pause position variable to make it move back to the original position. [This one](https://docs.octoprint.org/en/master/features/gcode_scripts.html#id5) is a good example of how this can be done.

If you have problems setting up these scripts to do what you like, please use the [community forum](https://community.octoprint.org) or the [discord server](https://discord.octoprint.org) for help configuring them. Thanks!

closed 2021-01-02 by cp2004

SysGh-st, 2021-01-02:
Thanks. for pointing me in the right direction.
I do wonder though. These scripts found under .octoprint/scripts/gcode/ . Are they generated/copied by some defaults when the printer profile is created? If so I suspect they're at fault then.
I was able to fix it by adding the missing Z-movement command to the beforePrintResumed script.

cp2004, 2021-01-02:
By default, there should be nothing in the pause/resume scripts. These are the default ones:

SysGh-st, 2021-01-02:
Huh.. okay... 
I'll have to check it out. I never created these files. I'll check if any of the plugins at some point might have created those scripts.