OctoPrint/OctoPrint #3926: Status screen shows >100% during print

awlred, 2021-01-05:
[image]
Minor issue, looks like it didn't pass the correct file size into the status pane resulting in a false %age calculation.
.gcode was sliced in Cura and then arc-welded using plug in. 

Print completed without issue

cp2004, 2021-01-05:
And you expect us to reproduce this how? Never seen this problem before, and you **must** include the full issue template. It's not for our entertainment, I seriously would not know where to start with only a screenshot to go by.

awlred, 2021-01-05:
> 
> 
> And you expect us to reproduce this how? Never seen this problem before, and you **must** include the full issue template. It's not for our entertainment, I seriously would not know where to start with only a screenshot to go by.

Apologies, I skipped the intro in haste. 

Reattempted in Safe Mode and it doesn't replicate so assumably is due to one of my plug-ins.