OctoPrint/OctoPrint #3927: [Request] Ability to easily tell OctoPrint to continuously try to reconnect the serial connection

JasonKeirstead, 2021-01-05:
Context: I have a Creately Ender 3. The Ender has a hardware external switch to turn the printer on and off, there is no software power off that shuts down the entire printer (including the PSU). As such, I have my Ender connected to a wifi smart outlet, which is connected to SmartThings. I have my printer also connected to SmartThings (using https://community.smartthings.com/t/octoprint-server-device-handler/188446), and have a simple automation set up so that 5 minutes after my print is complete, it shuts down the printer. 

This works great except for *one* very annoying thing - that the next day / whenever I power the printer back on, OctoPrint does not automatically restore the connection. I have to log in and click the "Connect" button every time.

I have read through issue #1388 - but have not been able to find any combination of the settings described that will cause OctoPrint to automatically reconnect when the serial connection has been removed and then re-added. I set all of the attempts to 604800 seconds and they still do not reconnect. This may be because I use windows for my OctoPrint server, I am unsure. 

I also tried installing the PortFinder plugin as it was suggested this may do an auto-reconnect, but alas it also did not help.

The workaround I have now is I am *continuously* calling "connect" from the REST API every minute, 24/7/365. This works,  but in my opinion is needlessly messy. It would be simpler to have an option to simply tell OctoPrint to reconnect to the serial port automatically when it shows back up.

cp2004, 2021-01-05:
Similar to #3846, probably better suited to a plugin than core OctoPrint, see reasons set out in https://github.com/OctoPrint/OctoPrint/issues/3846#issuecomment-738755163 specifically. Doesn't necessarily apply to all printers, and too high risk of breaking things was the response.

I'm also curious as to why the PortLister didn't work for you - it says it has an 'auto connect' option when the port re-appears, so if this does not work then it would be better to work out *why* it didn't work, and report it as an issue, than to implement it in OctoPrint directly stepping on an already well established plugin.

JasonKeirstead, 2021-01-05:
@cp2004 I suspect PortLister does not work for the same reasons in #3846 

I did not find #3846, I will just pile onto that one.

closed 2021-01-05 by JasonKeirstead