OctoPrint/OctoPrint #3930: Issue with octoprint and cura - Error 500 Internal Server Error

dwolfuk, 2021-01-06:
Raspberry Pi 4B 2G
Cura 4.8.0
OctoPrint 1.5.2
Python 3.7.3
OctoPi 0.18.0

Everything seems ok until i send something to print from cura. Shortly afterwards cura spams with 500 Internal server errors.

The following is repeated in the octoprint log files. This is a fresh install with a few plugins installed. Whilst cura shows these errors the camera feed continues uninterupted and the printer continues to warm and print. The eta, temperatures don't update in cura.

In the octoprint interface the uploaded file doesn't show and there's no details for the file yet it prints fine. after the file is uploaded the octoprint web interface stops reporting temperatures etc also.

Below is the log section thats repeated showing the error. Any suggestions would be appreciated :) - It all worked fine on the previous octopi/python image. Hopefully the error means something to somebody else i'll go back to the working setup.

Thanks

2021-01-06 10:02:04,599 - tornado.access - ERROR - 500 GET /api/job (::ffff:192.168.0.76) 7.77ms
2021-01-06 10:02:06,595 - octoprint - ERROR - Exception on /api/job [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/vendor/flask_principal.py", line 199, in _decorated
    rv = f(*args, **kw)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/job.py", line 86, in jobState
    "state": currentData["state"]["text"],
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/util/flask.py", line 309, in fixed_jsonify
    dumps(data, indent=indent, separators=separators, allow_nan=False) + "\n",
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/json/__init__.py", line 211, in dumps
    rv = _json.dumps(obj, **kwargs)
  File "/usr/lib/python3.7/json/__init__.py", line 238, in dumps
    **kw).encode(obj)
  File "/usr/lib/python3.7/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/usr/lib/python3.7/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
ValueError: Out of range float values are not JSON compliant

cp2004, 2021-01-07:
> This is a fresh install with a few plugins installed.

You need to make sure it is not a problem with the plugins then, and fill out the whole template. It's no good me trying some combination of the 269 plugins available to try and reproduce this.

dwolfuk, 2021-01-07:
thankyou for your response. I only had the psu control plugin and portlister plugins in use since i need them to control printer power. I did try safe mode and got the same result whilst manually applying voltage to the power relay.

I then tried a fresh install and only updated octoprint. I thought i'd cured it but after 1 seemingly fine print the error started again. It seems to be more a cura issue than octoprint but when it occurs the octoprint web interface freezes (no temps reported etc). Removing and re-adding the printer in cura appeared to cure it for a while aswell in that if i create a new printer there was no error but as soon as i switched back the error occured and this seemed repeatable. an hour later on the new created printer the error started so not sure.

I got frustrated and went the nuclear route. wiped everything and started again :)

Right now i've started with a nightly octopi 0.18 build, fresh install of cura 4.8.0 and its being printing for hours on different prints. I've added some plugins and it still appears ok. The jury is still out but if its ok in the morning then maybe using the latest nightly was the answer. I have a feeling the issue occurs after upgrading octoprint but i've no idea why. This nightly has the latest version of octoprint so thought it would eliminate that being a potential issue. I don't understand the error messages so i'm shooting in the dark. Just seems odd that 1 day everything is working and the next it isn't. If this doesn't cure it then i'll go back to octopi 0.17 and hopefully that will work :)

cp2004, 2021-01-07:
The point of a bug tracker is not necessarily for you to analyse the issue and work out what's wrong, it is for you to provide as much information as possible to help us find and fix the problem. If you don't understand the error messages, that is fine, we do since we've worked with OctoPrint extensively.

It is not a case of I don't want to help you, but if you want support you must use the [community forums](https://community.octoprint.org) or [discord server](https://discord.octoprint.org). If you want to report a bug, then you should fill in all the information requested in the [contribution guidelines](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md), and even when you have not done this ignore the bot asking you to do so. This is not for our entertainment, it is so we can actually get things done rather than wait around all day chasing information.

dwolfuk, 2021-01-07:
Its not often i ask for help and your responses only reinforce the reason why. I usually work things out myself but this had me stumped so i thought i would ask. In the end it appears to be resolved and even if it isn't i will work through it myself so i don't waste anymore of your time. I thought i had given enough information and the response from the bot suggested if i didn't provide more info then then i wouldn't get help. I didn't know there was a template until the bot responded. You say you understand the error messages yet you didn't give any kind of useful information as to what they meant or where to look. 

As it turns out running without plugins gave the same result. I fail to see what information you would need that wasn't provided but seems you just wanted to be picky and point out what i had done wrong which is neither helpful or encouraging. 

You could simply have ignored the issue if you didn't actually want to help. I wouldn't have been offended since the bot stated that the ticket would be closed in 2 weeks. It also stated that it was 2 weeks to provide information yet you comment the next day without giving the time for me to provide said information. If you're going to point out what others have done wrong then you need to be prepared for others to point out what you did wrong. I thank you for your time and in future wont report issues here

cp2004, 2021-01-07:
I apologise if the tone of these responses was not up to scratch, reading them again I can see exactly how you would interpret it this way. It was not intended this way, and I hope you can accept this apology. I was not sure if I should respond, but wanted to try and make something right here.

I spend way too long reading problems, on the community forums, discord server and bug tracker. Probably too long if I'm being honest. It is *incredibly* difficult to work out problems most of the time, and frequently requires back and forth with people which can be a *very* long process before we start to get anywhere useful. Many times, I get fed up of asking for the same information each time.

I wanted to type out a long explanation of why we have contributing guidelines here, that GitHub prompts you to read when opening an issue; why we create templates, so that when you press the 'new issue' button there is a bug report template already there too. We try and make things as easy as possible for you, so it is as easy as possible for us too. You don't want to read to me going on and on.

I do always give people 2 weeks to provide the requested information - before we deem it to no longer be an issue - and just try and point out the most important part of safe mode and log files occasionally. People often respond better to real people than bots, and feel more inclined to provide all the info. I got frustrated when you clearly had plenty of time to investigate the issue, and type a long reply, but there was nothing like a log file in sight, or anything that looked like concise steps to reproduce the problem that I could follow to work it out myself. Everything in the list of questions in the template are all things that provides important clues as to what had been going on behind the scenes.

We'll leave it at that, if you want to open more issues or seek support anywhere across the OctoPrint community, we will all (myself included) be there to help out, whatever the problem.

dwolfuk, 2021-01-23:
I finally got a couple of days to try and resolve this. I tried 3 different pis (3B, 4B 2gb, 4B 8gb), many different sd cards and usb drives, all octopi versions from 17 to 18rc and many inbetween with different versions of octoprint (32 and 64bit) do exactly the same I used virtual printer and uploaded the g-code directly to web interface to eliminate the printer or cura being the issue. On fresh installs each time i was banging my head against the wall. I went right back to the octopi 17 and my pi3b that had previously worked fine for years and faced the same issue.

Then i realised the only thing that was still common was the actual g-code. I wiped out my start g-code in cura and added back a line at a time. The line M150 W40 R20 G20 B20 appears to upset octoprint but i have no idea why. M150 W40R20G20B20 works fine.  I don't remember changing anything here but this is what seems to have fixed my issue for others that find themselves with the same. Don't forget to check the g-code lmao

I don't know why that line gives issue when theres a fair few similar lines at the start of my start g-code but its def appears to be. Its to control LED striplights. Very odd that even with the offending line the print works and its only when it stops that the issue appears. Very odd and its only been a few hours since i "fixed" it but seems to be repeatable. I can repeat this on all the different setups i tried on vanilla octopi images with no plugins so it should be easy for you to replicate yourself :)

foosel, 2021-01-23:
Could you please share a file that contains this problematic line? Considering that the error message you posted points at an encoding issue, it could very well be that there's something up in the line that doesn't show up as a printed character on screen.

dwolfuk, 2021-01-23:
[CCR20_xt90capstore.zip](https://github.com/OctoPrint/OctoPrint/files/5860526/CCR20_xt90capstore.zip)
thats the gcode file i used for my testing. line 44 seems to be the culprit. I can't see anything untoward but i've no idea what i'm looking for tbh. Removing that line (or for me editing it) seems to have cured it. only time will tell but if i add the spaces back on that line it seems to trigger the issue again. Its very odd as the same command (with differing values) elsewhere in the g-code dont present the issue. Could the G20 be presenting an issue if something is reading that as a seperate command? By removing spaces it must read differently somewhere, I have no idea lol. I'll report back if the issue returns but seems pretty repeatable in my setups.

[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/5860534/octoprint.log)
thats the octoprint log on a fresh install. I did also try in safe mode just incase it was a bundled plugin and the result was the same.

foosel, 2021-01-25:
I can reproduce this by just uploading the attached gcode file. The reason is that somehow the gcode analysis turns up an infinite z height of the model, and that causes invalid json on the model size report returned on file list & job requests.

So that is a bug.

Two things need to be solved here:

  * [ ] OctoPrint should not try to produce invalid JSON on the files API if there's garbage in the analysis reports - I thought that had already been ensured a while ago but apparently something is still able to get past the checks in place:
    ``` json
    "CCR20_xt90capstore.gcode": {
        "hash": "27998771a5ba6985b36f429c5e0b1631da4cb966",
        "analysis": {
            "printingArea": {
                "maxX": 70070.18243999999,
                "maxY": 70403.08499999999,
                "maxZ": null,
                "minX": 0.1,
                "minY": 20.0,
                "minZ": 0.3
            },
            "dimensions": {
                "depth": 70383.08499999999,
                "height": Infinity,
                "width": 70070.08243999998
            },
            "estimatedPrintTime": NaN,
            "filament": {
                "tool0": {
                    "length": 597.7630300000001,
                    "volume": 0.0
                }
            }
        }
    },
    ```
  * [ ] Something's causing the GCODE analysis to produce an infinit maxZ here, which is confusing considering that there are more than enough regular Z movement commands inside the model. `M150` is unknown to the analysis and *should* not influence the results but somehow does, because if I strip the line the analysis produces a valid result: 
    
    [image]

foosel, 2021-01-25:
Ok, so the issue is that the `G20` inside the `M150` there gets read as an `G20` *command* (parser error, fix ready). And replacing it with just a `G20` leads to the same broken behaviour, so something's up with `G20` vs the analysis.

dwolfuk, 2021-01-25:
Yes i figured the spaces caused the G20 to be read as a seperate command. In marlin G20 changes the units to inches but no idea if thats even relevant here as it would seem that change should take place on a printer rather than in octoprint. Its odd that its only the Z that changes though and that it still prints ok. Only when the prints finish or are cancelled things go south. I'm just pleased you can reproduce it. Most of the time thats the hardest part troubleshooting as a developer :)

foosel, 2021-01-25:
> Yes i figured the spaces caused the G20 to be read as a seperate command

Not per se, but that particular code location was buggy.

> Its odd that its only the Z that changes though and that it still prints ok

This has nothing to do with printing at all. It's only the file analysis that produced invalid results, and those made it into the metadata structure, and then caused internal errors on the API endpoints. Printing is completely out of the loop there.

I'm surprised it's only the z changes after identifying the issue tbh, but it's now fixed.

> Only when the prints finish or are cancelled things go south.

Go south where and how please? The only problem I'm seeing here is the HTTP errors caused by the invalid metadata.

dwolfuk, 2021-01-25:
The http errors/octoprint crashing are what i mean. Its only when the print is stopped or finishes that these errors occur. Ohhh i think i understand now. Its after the print when it tries to refresh and read the metadata on the file list that it fails. That might explain why i was still having issues there after i fixed the start g-code. wiping the uploads directory and restarting cured that which now makes sense if there was still something untoward there.  

Thankyou for your help though confirming and fixing the issue. It wasn't a fun few days trying to print and fight with the errors.

foosel, 2021-04-27:
1.6.0 has just been released.

closed 2021-04-27 by foosel