OctoPrint/OctoPrint #3931: Octoprint is reverting custom timelaps directory to default

Ajimaru, 2021-01-06:
#### What were you doing?
Octoprint is reverting custom timelapse directory (in my case **webmount/timelaps** ) back to default if custom directory is not avaiable.

#### What did you expect to happen?
Octoprint keep the custom directory.
For your understanding, the custom directory is a webdav share which is not instand avaiable after the octoprint server has finished booting, due to the Wifi connection of the Raspberry Pi 4.
Therefor I have a cron job which takes of that the webdav share is avaiable asap.

I can understand that Octoprint is checking the needed directories to ensure optimal functionality but in this case is make use of directories on different hosts difficult.

#### What happened instead?
Octoprint is reverting to the default directory **.octoprint/timelapse**

#### Did the same happen when running OctoPrint in safe mode?
Not tested but I am sure the behaviour is the same

#### Version of OctoPrint
1.5.1

#### Operating System running OctoPrint
Rasbian 10

#### Printer model & used firmware incl. version
Anet A6

#### Browser and version of browser, operating system running browser
Firefox 84.0.1

#### Link to octoprint.log
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/5776808/octoprint.log)

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

Custom timelaps directory
[image]

default directory after reboot
[image]

I have read the FAQ.
yes

foosel, 2021-01-11:
That is actually intended behaviour in order to avoid a non-working state due to broken folder settings (as has happened and severely upset users in the past). OctoPrint requires all folders you have configured to be there when it starts. In order to make your scenario work, you should make sure the startup script for OctoPrint depends on the file system for your timelapses already being available. For systemd, [this might be relevant](https://unix.stackexchange.com/a/247547).

Alternatively it might be an option to add a commandline flag to disable folder checking on startup. That's however something I'd put up for grabs and expect a PR for.

closed 2021-01-11 by foosel

Ajimaru, 2021-01-13:
Adding what described at the link to the octoprint systemd servcice did not helped.
Dont know what I can do to fix that except using the rpi internal memory to save the timelaps videos.

jneilliii, 2021-01-13:
just to clarify, this should probably be taken over to the forum, but did you run `systemctl list-units --type=mount` to find your custom mount, and then add what was listed or did you just copy/paste what was there for `After`?

Ajimaru, 2021-01-13:
Thats how the octoprint.service looks:

> [Unit]
> Description=The snappy web interface for your 3D printer
> After=network-online.target **home-pi-webmount.mount**
> Wants=network-online.target
> 
> [Service]
> Environment="LC_ALL=C.UTF-8"
> Environment="LANG=C.UTF-8"
> Type=simple
> User=pi
> ExecStart=/home/pi/OctoPrint/venv/bin/octoprint
> 
> [Install]
> WantedBy=multi-user.target

I'll take that over to the Octoprint forum.