OctoPrint/OctoPrint #3933: extra info button flickers

stefbrouwers, 2021-01-08:
#### What were you doing?
Looking what all the buttons do.

#### What did you expect to happen?
a box to pop up 

#### What happened instead?
a box popping up and closing repetedly causing flicker

#### Did the same happen when running OctoPrint in safe mode?

Not tested. But probly yes

#### Version of OctoPrint

OctoPrint 1.5.2  Python 2.7.16  OctoPi 0.17.0

#### Operating System running OctoPrint

OctoPi 0.17.0

#### Printer model & used firmware incl. version

Ender 3
1.1.6.2

#### Browser and version of browser, operating system running browser

Windows – google chrome - 87.0.4280.141

#### Link to octoprint.log

https://pastebin.com/s5YtMm2X

#### Link to contents of terminal tab or serial.log

[...]

#### Link to contents of Javascript console in the browser

?

#### Screenshot(s)/video(s) showing the problem:

https://user-images.githubusercontent.com/50795787/104070746-e418a080-5207-11eb-8dbe-9afee73f945b.mov

cp2004, 2021-01-08:
The only line missing is the 'I have read the FAQ' from the bottom, I'll give you a pass through the bot for now 🙂. It gets upset if *any* lines are removed when creating the issue, this might be when you inserted the video.

stefbrouwers, 2021-01-08:
> The only line missing is the 'I have read the FAQ' from the bottom, I'll give you a pass through the bot for now 🙂. It gets upset if _any_ lines are removed when creating the issue, this might be when you inserted the video.

thanks i was hella confused

cp2004, 2021-01-08:
I have reproduced this, though not consistently. It sometimes flickers, sometimes is perfectly normal. No logic to it, and haven't been able to get it back since.

I started fiddling with it and it worked just fine. Not entirely sure why this popover is using a custom KO binding, in place of the regular bootstrap data-* attributes. Maybe it is less characters.

foosel, 2021-01-27:
I'm pretty sure this is due to the popover being bound to hovering over the question mark symbol, and in your video it starts flickering right when your cursor enters its little anchor thing. Which would make sense - cursor on anchor means cursor no longer on icon, popover off, anchor gone, cursor at same position back on icon, popover up, rinse and repeat.

The thing though is:

[image]

for me the anchor doesn't reach down into the icon as it does for you. Window 10, Chrome Version 87.0.4280.141 (Official Build) (64-bit). So I cannot verify this theory. Could you check if this is indeed the trigger @stefbrouwers?

stefbrouwers, 2021-01-28:
yes @foosel , the trigger is when the info box obstructs the button deactivating the button makeing the box go away and then you get the flicker

foosel, 2021-02-01:
I fear that'll go onto the "won't fix" pile. I cannot for the life of me get it to the point to even open the popover overlayed like in your screenshot, and therefore cannot get it to do this. Even if I could though I would have to patch bootstrap in order to fix it, if it is even fixable at all (I'm not sure it is right now). I think I should probably rather spend the time that would eat up on more serious issues. However, if anyone wants to tackle this, I'd be happy to accept a PR.

closed 2021-02-01 by foosel