OctoPrint/OctoPrint #3934: Octoprint crashed after print finished

lhartman89, 2021-01-08:
#### What were you doing?
Print finished and I lost connection to the server. I could not SSH into the PI.

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

#### What did you expect to happen?
Server should still be up and running after print finishes
#### What happened instead?
PI completely crashed.
#### Did the same happen when running OctoPrint in safe mode?
Didn't try

#### Version of OctoPrint
Version 1.5.2

#### Operating System running OctoPrint
OctoPi

#### Printer model & used firmware incl. version
Ender 3 SKR E3 V2.0 running Klipper

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log
[https://pastebin.com/Tydqvb5Y](url)

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

cp2004, 2021-01-08:
> **Did the same happen when running OctoPrint in safe mode?**
> Didn't try

Please do so.

It doesn't appear OctoPrint itself crashed, or if it did it didn't log a single thing. More likely this was caused by an issue in the Raspberry Pi, for example the power cutting out. Useful log files for this issue can be found under `/var/log/`, such as the kern.log, the syslog.

Please make sure this is a bug with OctoPrint, it doesn't even seem to be related to the print in any way - there was 8 hours between the print finishing and the log cuts off. You have klipper running too, how do you know this is not related to any other software? We would like to know if you can pinpoint it to something OctoPrint has directly caused, there are thousands of reasons that OSes and Raspberry Pi's crash.

tedder, 2021-01-08:
Definitely needs to be run in safe mode- there are one or more failing plugins.

Once you do, post the logs to [community.octoprint.org](https://community.octoprint.org/) rather than here, unless you are reasonably confident there's a bug in Octoprint.

closed 2021-01-08 by tedder

lhartman89, 2021-01-08:
Charlie,

I wasn't blaming octoprint in particular and I didn't know where or how to
go about seeking help. But I was hoping my log might show something that I
was missing. It's has happened a couple of times, but not very often. I
figured maybe a plugin caused a crash or something.

Sincerely,
Lee Hartman