OctoPrint/OctoPrint #3936: [Request] Allow web-based update at beginning of set-up flow

4wrxb, 2021-01-10:
When setting up OctoPrint the wizard allows uploading a backup from an existing install. However, if the OctoPrint instillation is out of date (e.g. when using an OctoPi image) there is simply a "Backup is from a newer version of octoprint and cannot be applied" error from the console with no suggestion to resolve the issue.

The most user-friendly solution would be to provide an update function on this screen or at the beginning of he wizard, but even providing help steps similar to [these instructions](https://community.octoprint.org/t/how-can-i-update-the-octoprint-installation-on-my-octopi-image/207) (although perhaps more general if there are other installs where this could happen) would be a big help in avoiding user confusion.

cp2004, 2021-01-10:
Personally I think this is a very good idea, and maybe even forced? I suppose it might struggle if it's not connected to the internet, but that can be handled. Force an initial setup screen that checks for updates (see similar to how Home Assistant OS operates) and apply it. Lots of consumer products get shipped with out of date software that needs updates out of the box, this isn't some radical idea!

This would have two benefits that I can see:
* First being the part you mention about backups
* Second is that on older OctoPi images, such as OctoPi 0.17 it would mean we wouldn't get stuck with so many people on older software (see https://data.octoprint.org). It just means that any new users are automatically on new versions out of the box.

Maybe not force, but strongly recommend with a small button to skip with a confirmation dialog. So if people *really* want to roll back, they can. Absolutely forcing things sometimes doesn't sit well with a vocal minority.

foosel, 2021-07-07:
This has been implemented and will be in OctoPrint starting with 1.7.0.

I have decided against a VERY aggressive push for now (mandatory wizard step) and instead implemented it as the first step (even before backup restore) of the wizard with a strong suggestion.

Considering that in the future we'll hopefully have auto-updated OctoPi images thanks to [CustoPiZer](https://github.com/OctoPrint/CustoPiZer/) and [OctoPi-UpToDate](https://github.com/OctoPrint/OctoPi-UpToDate/) it should be less urgent than it used to be.