OctoPrint/OctoPrint #3937: Gcode viewer tab displays wrong layer number after gcode edit/restart

KenLucke, 2021-01-11:
#### What were you doing?
I had to restart a print after an accidental pause via OctoPod (I forgot to lock the controls, and it's really easy to accidentally hit something) which would not resume, which required editing of the gcode file to clear out the layers already printed.  When I restarted it, the gcode viewer tab info (below the viewer) started counting layers at zero, or one, or whatever it does  (Cura slices with the first layer numbered zero).  I waited a couple of layers to see if it would correct itself at layer change, but it didn't.

The "State" panel immediately displayed the correct layer number, even after the edit (i.e., I  restarted at layer 47, it properly displayed  "Layer 49" (because of the 2 layer difference between OP's shown layer and Cura's numbering).

1. Load an edited gcode file that doesn't start with the first layer
2. Start the print
3. Monitor the layer numbers in the gcode viewer and the "State" panel.

Here's the beginning of my file that I discovered this with:
```
;FLAVOR:Marlin
;TIME:48568
;Filament used: 45.0987m
;Layer height: 0.2
;MINX:63.942
;MINY:77.32
;MINZ:0.2
;MAXX:245.942
;MAXY:242.772
;MAXZ:20
;Generated with Cura_SteamEngine 4.8.0
M140 S65
M105
M190 S65
M104 S215
M105
M109 S215
M82 ;absolute extrusion mode

M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

G1 F2700 E-1
;LAYER_COUNT:100  <<<<<<<<<<<<<<<< True layer total

G0 F300 X194.275 Y130.915 Z9.6 <<<<<<----- next few lines setting the printer back up as it was when stopped
G0 F18000 X175.345 Y146.325
G0 X134.655 Y146.325
G0 X134.475 Y146.325
;TIME_ELAPSED:37592.438991

;LAYER:47    <------------------- New beginning layer
;TYPE:WALL-INNER
;MESH:Cross
G1 F2700 E1153.48726
G1 F3000 X115.525 Y146.325 E1154.11754
G1 X115.525 Y127.375 E1154.74782
G1 X134.475 Y127.375 E1155.3781
G1 X134.475 Y146.325 E1156.00838
G0 F18000 X134.875 Y146.725
;TYPE:WALL-OUTER
G1 F3000 X115.125 Y146.725 E1156.66526
G1 X115.125 Y126.975 E1157.32215
G1 X134.875 Y126.975 E1157.97904
G1 X134.875 Y146.725 E1158.63593
G0 F18000 X134.195 Y146.045
;TYPE:FILL
 : 
 : 
 : 
```

#### What did you expect to happen?
Should show correct layer

#### What happened instead?
Gcode viewer tab started counting layers, rather than displaying layer number encoded in file.

#### Did the same happen when running OctoPrint in safe mode?
Did not test, print in progress,

#### Version of OctoPrint
#### Operating System running OctoPrint
```
browser.user_agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:84.0) Gecko/20100101 Firefox/84.0
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 3927818240
env.os.bits: 64
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.3.3
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.2
printer.firmware: Marlin 2.0.7.5_NIC NICW VEADBB (Nov 23 2020

```

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to contents of terminal tab or serial.log
n/a

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

foosel, 2021-01-11:
The comments in GCODE are non standardized, heavily differ across slicers and slicer versions and thus cannot be relied on by OctoPrint to extract information from. The GCODE viewer behaves correctly here. The layer view in your printer state panel is provided by a plugin that apparently does parse the GCODE comments.

closed 2021-01-11 by foosel

jneilliii, 2021-01-11:
> The layer view in your printer state panel is provided by a plugin that apparently does parse the GCODE comments.

I think this is Display Layer Progress to be exact, not core OctoPrint functionality.