OctoPrint/OctoPrint #3938: Octoprint briefly flashes "Server is currently offline. Reconnect..." when logging in

timur-tabi, 2021-01-11:
#### What were you doing?

Whenever I log in, the Octoprint login screen briefly (fraction of a second) shows the error message, "Server is currently offline. Reconnect..." before successfully logging in.  I suppose technically this is a cosmetic issue, but it is disconcerting and should be fixed.

Problem also occurs in safe mode.

OctoPrint 1.5.2
Python 2.7.13

OctoPi 0.15.1

Prusa MK3s with latest firmware

Firefox 84.0.2 (64-bit) running on Kubuntu 20.04.

[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/5798347/octoprint.log)

[console-export-2021-1-11_15-12-43.txt](https://github.com/OctoPrint/OctoPrint/files/5798370/console-export-2021-1-11_15-12-43.txt)

[image]

I have read the FAQ.

jasonbcox, 2021-01-13:
I've encountered this as well. It's jarring. A quick look at the code and it looks like the socket is kicking off the onDisconnect callback erroneously (but probably harmlessly) in some cases.

foosel, 2021-02-09:
Fix should be ready for 1.6.0

commit 0904f991 (2021-02-09): 🐛 Login: ignore disconnect on page reload

Closes #3938
closed 2021-04-27 by foosel