OctoPrint/OctoPrint #3939: [Request] Add M999 to list of emergency commands

mrseeker, 2021-01-12:
**Is your feature request related to a problem? Please describe.**
When the BLTouch stops probing, it results in an error that needs to be unblocked by sending the command "M999". Unfortunately, this command cannot be sent when the "busy" protocol blocks any commands, and thus it results in a forced disconnect from the printer that can only be resolved by either waiting until the firmware replies to M110 or when it's reset.

**Describe the solution you'd like**
The possibility to send M999 (and other emergency commands), regardless of state.

**Describe alternatives you've considered**
Every time the firmware locks up, reset the printer manually.

foosel, 2021-01-12:
Is `M999` part of the emergency commands by default in Marlin and friends?

mrseeker, 2021-01-12:
It's not an "emergency" command as in "halt the printer", but more a "reset" command when the firmware encounters an error and requires human intervention to solve the issue (eg probe getting stuck):
https://reprap.org/wiki/G-code#M999:_Restart_after_being_stopped_by_error

It's similar to M108 in the sense that it continues after a reset...

foosel, 2021-01-12:
You misunderstood my question. I know what `M999` does. The question is whether it is supported by the emergency parser on firmware side, as in, there's actually firmware support to allow this command to "jump the queue" so to speak. Current documentation at least in Marlin indicates it's actually not:

https://github.com/MarlinFirmware/Marlin/blob/1314b31d97bba8cd74c6625c47176d4692f57790/Marlin/Configuration_adv.h#L975-L979

And without there being support for that, it doesn't make sense at all to declare it as an emergency command on OctoPrint's side. Emergency here doesn't relate to the context in which that command gets used, but rather is a special type of command which are parsed in a different way to be executed as early as often and which do not need to adhere to the general ping-pong requirements of the communication protocol.

mrseeker, 2021-01-12:
Upon further investigation, it seems that the "STOP" command actually does not stop the Marlin firmware from processing further lines, but rather stops the BLTouch from working until it receives the reset command. By sending the M999 command, octoprint in return will try to send the message and disconnects soon after ("no reply from firmware"), but the end result is that Octoprint can't connect until the running cycle is complete (after which it tries to unblock itself)... Will file bug with Marlin because I think this is an "unintended feature" due to the occurrence of a blocking race condition (BLtouch requires M999 to reset but underlying G29 command does not halt).

closed 2021-01-12 by mrseeker