OctoPrint/OctoPrint #3940: Octoprint sends M25 when cancelling SD print

MarkMan0, 2021-01-12:
#### What were you doing?
Printing from SD card.
Using Marlin version 2.0.5 and latest Octoprint on a RPI 3. I've launched an SD print, and when I tried to cancel it, it went into pause - both using the LCD and octoprint.

Now, I believe the issue is, that Octoprint sends `M25` when clicking Cancel, and this puts the firmware into Pause state. Octoprint should be sending gcode `M524`, which is for aborting an SD print.

Can I change it somewhere in the config files, I couldn't find any scripts which are executed when clicking Cancel button, maybe I overlooked it.

After Clicking on Continue in the prompts, Octoprint sends `M26 S0`, which resets the file, but it doesn't close it, so the printer starts again from the beginning.

In the serial You will see `Recv: SD paused`, this is a new SD state I introduced, because without this, pause would not work. After clicking Pause, the printer would send "Not SD printing", and Octoprint would then cancel the print.

I realize these compatibility issues are hard to solve, because both parties need to change their software, but is there a way for me to fix this for my printer? I like to print from the SD card, but I'd also like to remotely pause and cancel the prints.

Where could I change the gcodes executed when clicking Cancel?

1. Connect Octoprint
2. Start an SD print
3. Wait after all the heating is done and the printer is actually printing
4. Cancel the print via the LCD or Octoprint - same results
5. Printer goes into pause
6. Go through the prompts to continue printing
7. Printer restarts the file

#### What did you expect to happen?
I expected the printer to stop printing, like when cancelling w/o Octoprint
#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?
Yes, it did.

#### Version of OctoPrint
Octoprint 1.5.2
Python 2.7.16
OctoPi 0.17.0

#### Operating System running OctoPrint
Downloaded from octoprint.org

#### Printer model & used firmware incl. version
Marlin 2.0.5

#### Browser and version of browser, operating system running browser
Firefox 84.0.2

#### Link to octoprint.log
https://gist.github.com/MarkMan0/b5917d9e7275376516b54f9ecbd21931

#### Link to contents of terminal tab or serial.log
https://gist.github.com/MarkMan0/905f20ade98803257b6f722da4059026
Cancel clicked around line **138**, `M26 S0` sent on line **244**

#### Link to contents of Javascript console in the browser
I don't think it's needed.

I have read the FAQ.

foosel, 2021-01-12:
That is intended. There's no command to cancel an SD print defined in the protocol, so all that OctoPrint can do is pause it and simply not offer the UI option to restart it. My hands are sadly tied here by what's available on the serial interface 🤷‍♀️

closed 2021-01-12 by foosel

bcat, 2021-09-17:
I'm not super familiar with Marlin or GCode more generally, but it looks like there is an [M524 - Abort SD Print](https://marlinfw.org/docs/gcode/M524.html) command now. Is that something reasonable for OctoPrint to support? As someone who usually does SD prints, it would be nice for OctoPrint to fully support remote cancellation there.

Alternatively, are the pause/cancel operation something plugins can hook into? If so, conceivably a plugin could be developed that uses M524 if and only if an SD print specifically is in progress.

foosel, 2021-09-17:
It depends. If there's a way to determine that the firmware on the other hand supports M524 (read: there's a capability flag), then yes, support for this can be added to core, replacing M25 with M524 for cancelling in such a case. But if not, OctoPrint has to rely on the least common denominator among firmware variants out there, and that is M25.

A plugin could easily step in in such a case, and [rewrite](https://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-comm-protocol-gcode-phase) M25 to M524 if containing the tag `trigger:cancel` that the core code attaches on sending.