OctoPrint/OctoPrint #3945: Octoprint 1.5.2 "Internal Server Error" when connecting through one of the computers inside the home network

EagleB3, 2021-01-13:
#### What were you doing?

OctoPi Version 0.17.0, running on Raspberry Pi 3 Model B Plus Rev 1.3 + Python 3.7.3 + Octoprint Version 1.5.2. 
Raspberry is connected to a home router via WiFI, it is always assigned a fixed address 192.168.1.44.
Octoprint restarted **in safe mode** (In normal mode, the problem is the same).
I have a normal connection and work through my Desktop computer (connected to my home network via a TP cable, it is always assigned a fixed address 192.168.1.12).
I get "Internal Server Error" when trying to connect from laptop (connected to my home network via WiFi, it is always assigned a fixed address 192.168.1.51). 

In the **octoprint.log** I see the phrase "netaddr.core.AddrFormatError: **invalid prefix for IPv4 address**!"

#### What did you expect to happen?

I expected there would be no problems. They never existed when running Octoprint 1.4.2 and earlier.

#### What happened instead?

I am getting a problem logging in from laptop. Laptop equipment, settings of its network adapter (IP v.4, IP v.6), the parameters of the connection to the network and the home router did not change.
There is still no problem signing in with Desktop.

#### Did the same happen when running OctoPrint in safe mode?

YES

#### Version of OctoPrint

Version 1.5.2.

#### Operating System running OctoPrint

OctoPi Version 0.17.0, running on Raspberry Pi 3 Model B Plus Rev 1.3 + Python 3.7.3

#### Printer model & used firmware incl. version

Smoothieware v.1,  in problem unaffected

#### Browser and version of browser, operating system running browser

Windows 10 64-bit, Firefox 84.0.2 x64

#### Link to octoprint.log

[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/5809230/octoprint.log)

Up to line 142, the connection from the desktop is logged (192.168.1.12)
Starting from line 142 (2021-01-13 13:36:55), the process of connecting a laptop begins.

#### Link to contents of terminal tab or serial.log

Serial interface in problem unaffected

#### Link to contents of Javascript console in the browser

[image]

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

cp2004, 2021-01-13:
This doesn't appears to be an issue the Auto Login Local configuration you have setup.

Please try removing or disabling this section in config.yaml and restart the server. I will have to go and check if `192.168.1.0/64` is valid, unless someone smarter gets here first.

foosel, 2021-01-13:
IPv4 addresses are 32bit max, so `/64` makes absolutely no sense there. For 192.168.1.x it should probably be `/24` instead.

cp2004, 2021-01-13:
I should have googled it before commenting, it wasn't hard to find it...

foosel, 2021-01-13:
Might make sense to add some validation in OctoPrint to make sure the subnet classifiers are actually valid...

EagleB3, 2021-01-13:
The router has a pool of home network addresses that can be assigned to connections (192.168.1.2, 192.168.1.3 .. 192.168.1.254) . I do not assign a fixed network address allocation to all of my devices, but only to some devices. For them, the address will be allocated dynamically. And the "Papa" user should be able to connect to octoprint from any device on the home network.

I read the manual "How to set up OctoPrint to autologin a single user when connecting from the internal network" as soon as it appeared for version 1.5.0. But I didn’t find (or didn’t understand...) how to allow the user to connect from **any home network** address, 192.168.1.2 ... 192.168.1.254.

I was guided by the example of setting the localhost address **range** given there:
[image]

I tried to allow connection from addresses **range** 192.168.1.0 ... 192.168.1.64, specifying "yourAddressRange" = "192.168.1.0/64" in the settings. 
If I set the address **range** incorrectly, how should I do it correctly?

foosel, 2021-01-13:
> how should I do it correctly?

`192.168.1.0/24`. The 24 here means "these many bits of the thing before the / are fixed, the rest is dynamic" - that captures `192.168.1` as fixed and leaves the fourth byte as a wildcard, matching anything from 0 to 255.

`192.168.1.0/64` says "take the first 64 bits of this as fixed". An IPv4 address only HAS 32 bits, so it makes no sense at all and gets denied. Set it to `192.168.1.0/24`, that's correct for your network definition as far was I can see (192.168.1.1 - 192.168.1.254, with gateway at 0 and broadcast at 255).

Any further segmentation needs to be done on the network level and is outside of the scope of OctoPrint. I'm closing this since it's clearly a configuration error that's causing you grieve here, and that should be fixed if you set the above.

closed 2021-01-13 by foosel

EagleB3, 2021-01-13:
I changed the value "192.168.1.0/64" in config.yaml to "192.168.1.0/8" and the problem was fixed.

[image]

THANK YOU VERY MUCH for explaining to me how ".0/N" notation works in the IP-addressing structure!

P.S. I have worked in the field of IT, software development and radio electronic devices for 30 years, but about network addressing and routing, I only know that it is there and a little more. But you guessed it.

foosel, 2021-01-13:
With /8 you've now told OctoPrint that anything that starts with 192. is local. That's wrong. At the very least set it to /16 (which reflects the class B network that your IP range is specified for), otherwise this is insecure. Your router setup certainly sounds like /24 would be correct.

EagleB3, 2021-01-13:
@foosel , Thank you very much! I set in the settings "192.168.1.0/24"