OctoPrint/OctoPrint #3948: Logging settings don't clear when "Close" is pressed instead of "Save"

eyal0, 2021-01-13:
#### What were you doing?

1. Open settings dialog
2. Go to "Logging".
3. Add a few rows in the "Logging Levels" section by clicking on the blue plus sign.
4. Click "Close" and not "Save".
5. Preform an action and notice that the logging was actually not affected because I didn't hit Save (good!)
6. Re-open the settings dialog.
7. Select logging again.
8. Noticed that the logging level that I set is still listed there, even though I never saved it.

#### What did you expect to happen?

I expected the settings dialog for Logging Levels to return to the previous values.  This is what happens for other settings such as the checkboxes in the "Features" section of the settings.

#### What happened instead?

The logging levels are still there, making it look like I saved when I didn't.  (The behavior is correct in that the the setting didn't take effect and logging was unchanged!  Just the appearance is wrong.)

#### Did the same happen when running OctoPrint in safe mode?

I didn't test it but I can't imagine that it would matter.

#### Version of OctoPrint

OctoPrint 1.5.2 Python 3.7.3

#### Operating System running OctoPrint

Linux

#### Printer model & used firmware incl. version

Virtual Printer.  I don't have a printer

#### Browser and version of browser, operating system running browser

Google Chrome

#### Link to octoprint.log

During the whole testing process, no log lines were created.

#### Link to contents of terminal tab or serial.log

Nothing.  No printing was done.

#### Link to contents of Javascript console in the browser

Nothing

I have read the FAQ.

eyal0, 2021-01-13:
This is related to #3947

eyal0, 2021-01-13:
I'll work on this.

foosel, 2021-01-18:
Thanks for the PR! It's merged and will be part of 1.6.0!

commit f8d8af8c (2021-01-18): Reload logging settings when settings are viewed.

This fixes #3948
closed 2021-04-27 by foosel