OctoPrint/OctoPrint #3950: Logging level for "octoprint" appears on Logging levels even though it isn't been logged

eyal0, 2021-01-13:
#### What were you doing?

1. Click on settings.
2. Click on "Logging"
3. See this:

[image]

It appears as if I have debug logging set for both `octoprint.events.fire` and `octoprint`.  However, only the first one is set.  The second one hasn't yet been added.

#### What did you expect to happen?

I expected only `octoprint.events.fire` to be listed because that's the only one that I have set.

#### What happened instead?

I saw `octoprint` listed for `DEBUG` even though I didn't enable that logging.

#### Version of OctoPrint

OctoPrint 1.6.0.dev178+g498463f5c Python 3.7.3

#### Operating System running OctoPrint

Linux

#### Browser and version of browser, operating system running browser

Chrome

I have read the FAQ.

eyal0, 2021-01-13:
This is related to #3949 
This is also related to #3948

eyal0, 2021-01-13:
I will fix this.

foosel, 2021-01-18:
Thanks for the PR! It's merged and will be part of 1.6.0!

commit 85919a1d (2021-01-18): Add empty component to availableLoggers list

This fixes #3950
closed 2021-04-27 by foosel