OctoPrint/OctoPrint #3952: Logging levels are saved even when there are no changes

eyal0, 2021-01-13:
#### What were you doing?

1. Open settings
2. Click on "Logging"
3. Add or modify any logging level (if you add, don't forget to click the blue plus sign!)
4. Click "Save" to save the changes.
5. Observe that `ConfiguredLoggers has changed. Saving!` is in the console log.
6. Open the settings dialog again.
7. Without changing anything, click "Save" again.

#### What did you expect to happen?

I expected not to see `ConfiguredLoggers has changed. Saving!` in the console log.

#### What happened instead?

I saw `ConfiguredLoggers has changed. Saving!` in the console log.

#### Version of OctoPrint

OctoPrint 1.6.0.dev178+g498463f5c Python 3.7.3

#### Operating System running OctoPrint

Linux

#### Browser and version of browser, operating system running browser

Chrome

I have read the FAQ.

eyal0, 2021-01-13:
I will fix this.

foosel, 2021-01-18:
Thanks for the PR! It's merged and will be part of 1.6.0!

commit 56c516aa (2021-01-18): Reset changed flag after reloading the settings.

This fixes #3952
closed 2021-04-27 by foosel