OctoPrint/OctoPrint #3954: Logging levels can be added more than once

eyal0, 2021-01-13:
#### What were you doing?

1. Configure at least one logging level in Setting -> Logging.  Don't forget to press the blue plus sign.
2. Click on "Save"
3. Reload Octoprint by pressing F5 in the browser.
4. Open settings and go to the Logging settings
5. Notice that it is possible to add the same logging level that already exists in the list.

#### What did you expect to happen?

It should not be possible to re-add a logging level

#### What happened instead?

It was possible to re-add a logging level

#### Version of OctoPrint

OctoPrint 1.6.0.dev178+g498463f5c.dirty Python 3.7.3

I have read the FAQ.

foosel, 2021-01-18:
Thanks for the PR! It's merged and will be part of 1.6.0!

foosel, 2021-04-27:
1.6.0 has just been released.

closed 2021-04-27 by foosel