OctoPrint/OctoPrint #3955: It's possible to add the same logging level twice

eyal0, 2021-01-13:
#### What were you doing?

1. Configure at least one logging level in Setting -> Logging.  Don't forget to press the blue plus sign.
2. Click on "Save"
3. Reload Octoprint by pressing F5 in the browser.
4. Open settings and go to the Logging settings
5. Notice that it is possible to add the same logging level that already exists in the list.

#### What did you expect to happen?

It should not be possible to re-add a logging level

#### What happened instead?

It was possible to re-add a logging level

#### Version of OctoPrint

OctoPrint 1.6.0.dev178+g498463f5c.dirty Python 3.7.3

I have read the FAQ.

eyal0, 2021-01-13:
I will fix this.

cp2004, 2021-01-13:
Is this a duplicate, or is #3954 a duplicate? Either way they are about the same issue, so one should be closed as far as I see.

eyal0, 2021-01-13:
It's a duplicate.  Github wigged out on my when I tried to save it the first time.  Sorry.

This is a duplicate of #3954

commit bf3461f9 (2021-01-18): Use _.difference instead of _.without for arrays

This fixes #3955
closed 2021-01-13 by eyal0