OctoPrint/OctoPrint #3959: Slow print speed with stuttering, random pauses (v1.5.2)

deviant-studio, 2021-01-14:
#### What were you doing?

Something really bad happens on latest Octoprint versions. I guess it started from 1.5.x. My prints take x2/x3 times longer than via sd card or via Octoprint v1.3.12 (which i currently use)

The printer head stutters and sometimes totally stops. I didn't notice any suspect things in the logs though. Tried 1.5.2 in safe mode with no any positive improvements.

Also tried several different usb cables.

I'm on Orange Pi Zero.
First i thought my hardware is dying. I swapped sd card with no positive result. Then i found my old backup dump (v1.3.12). I gave it a chance and voila! It works like a charm with appropriate speed and no blobs and pauses.

#### What did you expect to happen?

Should print not slower than 15% from what sd card print gives. (My expirience from v1.3.12)

#### What happened instead?

It prints 2-3 times slower. Especially when model have lot of curves.
Of course my prints were full of blobs due to constant stuttering.

#### Did the same happen when running OctoPrint in safe mode?

YES

#### Version of OctoPrint

1.5.2

#### Operating System running OctoPrint

Armbian on the Orange Pi Zero

#### Printer model & used firmware incl. version

Ender 3 Pro

#### Browser and version of browser, operating system running browser

Firefox on Win10

#### Link to octoprint.log

https://pastebin.com/HAKbdFkN

#### Link to contents of terminal tab or serial.log

https://pastebin.com/dgVA0180

I have read the FAQ.

gzahl, 2021-01-16:
I have the same issue, Octoprint running on Armbian, Banana Pi M2U. I tried in safe mode, reinstalled octoprint, without any plugins or only with OctoKlipper. Still always to CPU cores are at 100% and printing is so slow, it is not usable.
It was fine last weekend. Not sure what changed since then.

deviant-studio, 2021-01-16:
mine CPU load is within appropriate values though (5-10%). i suspect something wrong with serial port api

3dRikal, 2021-01-19:
Having the same issue. Small prints, especially ones that have tight circular motions are jerky and jittery. When i print from the SD card, the printing is fairly smooth (a lot smoother than octoprint). On Anycubic Mega S, Marlin 2.0, Standard drivers, USB Baud at 250000.

cp2004, 2021-01-19:
@3dRikal Please confirm as the previous commenters have that this issue **only** impacts OctoPrint 1.5.x. This is not going to be yet another thread about stuttering printing via USB, those can be found [here](https://community.octoprint.org/t/the-ongoing-usb-conspiracy-theory/6341), [here](https://github.com/OctoPrint/OctoPrint/issues/450) and [all over the forums](https://community.octoprint.org/search?q=stuttering). If the stuttering goes away with OctoPrint 1.4.2 then come back here but if it remains the same then this is not the same issue.

3dRikal, 2021-01-19:
I am new to octoprint and the only version I have used is 1.5.2. I was not in the loop for 1.4.x sorry. I will update to 1.5.3 today and am still conducting tests to eliminate some culprits. I am trying to determine if serial buffer from FW is at fault or usb cable is defective but so far, only prints from octoprint are failing (and only on tight circular prints)

3dRikal, 2021-01-19:
Tested with 1.4.2 and jitter is reduced to almost nothing compared to 1.5.2.

foosel, 2021-01-19:
In that case, test in safe mode, and if the issue persists please also fill out the [ticket template](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report).

foosel, 2021-01-19:
@deviant-studio that does not look like a full `serial.log` in your post.

For reference, the only thing change about the comm layer that I could see causing a bit more computational overhead is counting of resend requests vs total sent lines ([see history]). The ratio only gets calculated on resend requests however. I cannot reproduce a problem myself here so far, so whatever is happening might be very environment specific.

deviant-studio, 2021-01-19:
yes, the serial.log became really big, so my browser crashes when i copy/paste it into pastebin 🤷‍♂️ but nothing interesting there anyway. just repeating Recv: ok lines. no errors/warnings.
is there any way to stress test the serial connection? maybe some integration test that benchmarks the throughput?
also is it hard to build octoprint from the sources? i could check different branches and confirm if it's really due to last commits or some other (hardware specific) issue.

3dRikal, 2021-01-19:
i'll have to revisit this over the weekend. Got an order I need to push through. SD card will have to work for now.

foosel, 2021-01-20:
> is there any way to stress test the serial connection? maybe some integration test that benchmarks the throughput?

If you have an SD card in your printer, yes, in a way. You can use `src/octoprint/util/comm.py` to stream a file to the printer's SD and benchmark the transmission speed. However, that is more for measuring the general throughput capabilities of the general serial implementation, as in SD streaming mode a lot of the plugin hooks and such are bypassed.

Still, if you want to play around with it:

    python -m octoprint.util.comm <port> <baudrate> <local path> <remote path>

(in the OctoPrint venv, so on OctoPi that needs to be either `~/oprint/bin/python ...` or you need to activate the venv first via `source ~/oprint/bin/activate`)

> also is it hard to build octoprint from the sources? i could check different branches and confirm if it's really due to last commits or some other (hardware specific) issue.

Nope, not at all. There's no actually building involved really, just having a working Python install, checking out the sources, installing the dependencies and done. There's also a script on OctoPi to do all this for easy experimenting, see the README in `~/OctoPrint`.

deviant-studio, 2021-01-24:
i made some additional research and wanted to add more details:
i tried to upgrade 1.3.12 -> 1.5.2 (with web interface update) to see what happens. and again i got jerks.
then i tried to downgrade to 1.4.2 with pip install = jerks
then downgrade to 1.3.12 = jerks.
so probably not a octoprint issue but some dependency that installs together with 1.5.x

also i tried the arc welder plugin. it helps to get rid of jerks, but random pauses still occurs.
so i'm going back to sd card printing at least until i get some more powerful hardware.

nasp2000, 2021-02-06:
I have the same problem

nasp2000, 2021-02-21:
Well i have updated the orangepi zero for Next (Armbian 21.02.2 Buster with Linux 5.10.16-sunxi) and change the cpu from ondemand to performance. 

Also i have reinstall Octoprint now running with python3 and the problems off randam pauses disapear.

gloomytree, 2021-08-14:
Does this still persist in 1.6?

cp2004, 2021-09-22:
Could you confirm if you are using Python 2 vs 3?

Shambler2, 2021-09-22:
From System Info: 
env.plugins.pi_support.model: Raspberry Pi 3 Model B Plus Rev 1.3
env.plugins.pi_support.octopi_version: 0.15.0
env.plugins.pi_support.throttle_state: 0x50005
env.python.pip: 10.0.1
env.python.version: 2.7.13
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.6.1

cp2004, 2021-09-22:
> env.plugins.pi_support.throttle_state: 0x50005

Fix this first. Your Pi is throttling due to undervoltage issues, it can't get enough power so it's limiting what it will do.

Shambler2, 2021-09-22:
Thanks! Will check it out. Wasn't sure if that had much of an impact on prints - haven't changed a thing as far as powering the Pi in near 2 years without problems like this. Will run through a quick test with a beefier power supply and let you know

foosel, 2021-11-10:
> Will run through a quick test with a beefier power supply and let you know

@Shambler2 did anything ever come out of this test?

Shambler2, 2022-02-05:
Apologies for the rather ridiculous delay - I've finally resolved the power problem and come back to this issue (to be honest it turned me off using the printer as I just couldn't resolve it at the time.

Unfortunately, its still happeneing. I've completely re-installed OctoPrint and am now running the latest release, and Python 3. When I did this however I DID export, then re-import my settings.

Happy to do some more troubleshooting on this now if anyone is able to help!

Shambler2, 2022-02-06:
So after a lot of troubleshooting I've finally worked out that the issue was something in my Cura settings when slicing the STL - not sure what setting it was yet but I can see a 1MB file size difference in the gcode for the same object, I suspect I was just throwing too much detail into the file after all.

Dephrilibrium, 2022-07-16:
#### What were you doing?
as described

#### What did you expect to happen?
normal printing

#### What happened instead?
stuttering and stops (especially in curves with a lot of position data)

(Printing from SD-Card works fine)

#### Did the same happen when running OctoPrint in safe mode?
same

#### Version of OctoPrint
See System Info Bundle

#### Printer model & used firmware incl. version
MK3S (FW3.11.0-4955) +MMU2 (FW: 1.0.6-372)
(should be latest to the current date)

#### Browser and version of browser, operating system running browser
See System Info Bundle

#### System Info Bundle
browser.user_agent: Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:102.0) Gecko/20100101 Firefox/102.0
connectivity.connection_check: 1.1.1.1:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1200
env.hardware.ram: 914006016
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 3 Model B Rev 1.2
env.plugins.pi_support.throttle_check_enabled: true
env.plugins.pi_support.throttle_check_functional: true
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 22.1.2
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.last_safe_mode.date: unknown
octoprint.last_safe_mode.reason: unknown
octoprint.safe_mode: false
octoprint.version: 1.8.1
printer.firmware: Prusa-Firmware 3.11.0 based on Marlin
systeminfo.generated: 2022-07-16T21:42:45Z
systeminfo.generator: systemapi

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:
https://user-images.githubusercontent.com/89015665/179373563-f8b221c9-7b49-44ac-9335-5e05de003e96.mp4

Hope this helps a bit.

Kind regards,
Deph

AapoTahkola, 2022-09-23:
I had at least on outset similar looking situation with stuttering prints and octoprint starting from browser if at all. After much digging, I realized I had two mac addresses on orangepi pc one for eth and one for wifi. I disabled the wifi adapter on orangepi and wired eth to openwrt router. Then removed all dhcp static leases from openwrt that pointed to octoprint, reboot and assigned static dhcp lease to the eth. Finally reset cura octoprint plugin. Another thing that I suspected was unwanted process scheduling set by armbian-config cpu frequency and governor set too high of a setting.

Hello1024, 2023-12-18:
For anyone finding this issue later...    Upgrading python from 3.7 to 3.12 seems to help substantially.

It isn't easy to do though - I built 3.12 from source, and then had to mod this script to run python3.12 instead of python3: https://github.com/cp2004/OctoPrint-Upgrade-To-Py3

foosel, 2024-08-19:
Closing as this issue was explicitly about 1.5.2 and that's long outdated.

For general issues, see #450.

closed 2024-08-19 by foosel