OctoPrint/OctoPrint #3961: Octoprint service not starting anymore

JustAnother1, 2021-01-16:
#### What were you doing?

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

I powered on the Raspberry Pi3.

#### What did you expect to happen?

I expectet to be able to reach the Octoprint Web Interface.

#### What happened instead?

I got a page informing me that Octopi is not running an dthat I should investigate using SSH.
No Python process was running

#### Did the same happen when running OctoPrint in safe mode?

Yes same effect.

#### Version of OctoPrint

OctoPrint 1.4.2

#### Operating System running OctoPrint
Octopi (Linux)

#### Printer model & used firmware incl. version
Ultimaker / Marlin 1.?

#### Browser and version of browser, operating system running browser
na

#### Link to octoprint.log

https://gist.github.com/JustAnother1/ee36f78e648d8830435e84635f2f1afb

#### Link to contents of terminal tab or serial.log
na

#### Link to contents of Javascript console in the browser
na

#### Screenshot(s)/video(s) showing the problem:
na

I have read the FAQ.

JustAnother1, 2021-01-16:
From the log thsi seems to be a prominent issue:

ImportError: cannot import name flask

cp2004, 2021-01-16:
Your installation appears to be corrupted. This can be caused by unplugging the power from the Pi without shutting it down or a bad SD card.

You can try reinstalling OctoPrint, and see if that fixes the corruption issue. If it doesn't, please seek support on the [community forums](https://community.octoprint.org) or the [discord server](https://discord.octoprint.org). Please also see the contribution guidelines. Thanks

closed 2021-01-16 by cp2004

JustAnother1, 2021-01-16:
That is a cool way of looking at issues. So if octoprint ever has an issue, in the updater or some other component, that causes a corruption in the octoprint files then a new install will fix that also,...

With ext4 power down during operation is automatically fixed on boot. The corruption of files due to sudden power down also only affected files that were currently written to. So the octoprint files should not be affected by something like that. It could crash time-lapse files or uploaded g-code files or log files though.

But a new install fixes it so better not dig to deep one can only find hard work,....

foosel, 2021-01-16:
> so better not dig to deep one can only find hard work

Because OSS devs are known for being lazy bums who don't want to lift a single finger to help others and solve issues they don't experience themselves, and are shy of work, be it hard, repetitive or easy in nature. Yeah. Right.

Experience in the community has shown these issues creeping up from time to time for people due to the reasons @cp2004 mentioned. Take that advice or leave it, but leave the attitude at the door please.

cp2004, 2021-01-16:
> So if octoprint ever has an issue, in the updater or some other component,

Then you'll come to us saying 'i pressed update, it died', or 'i did X now it won't boot' rather than just power the Pi on.

It really does happen, whether you think it _should_ or not this community is quite large and we see these issues on a regular basis, out of the blue with no explanation. There comes a point where it is just instinctive experience, for so many things to fail in your install tells me something is very broken, out of the blue - all signs point to corruption. If you want to offer up an alternative explanation, by all means go for it and the issue can be reopened with one click.

JustAnother1, 2021-01-17:
I have been misunderstood and it is probably my error. I apologise!
I tried to give it a fun spin, but that went horribly wrong. Again my error and I'm sorry.

@foosel I did not intend to imply anything of what you understood.

All I intended and still intend is to not ignore this issue. Once it happens it is really annoying. And event if this is probably caused by a bad sd card or an issue outside of octoprint it IMHO should not be overlooked. Especially if as cp2004 says this happens "on a regular basis".

I don't expect anyone to write a fix for me. I know a bit of Python myself, so if I can help I would. But the Web stack with Java Script is too complex for me. I do more close to the hardware safety critical real time stuff. But if I can help by testing something then please let me know.

@cp2004 All I did yesterday was trying to power it on. I sometimes do not use the printer for longer times. Therefore I'm not surprised to see several update notices coming up on every boot. So I may have done an update the last time I used the printer. I don't know for sure. It then worked still after the upgrade. I did not reboot it to see if it still works after the reboot. So maybe that update caused something that then only came into effect on the boot yesterday. I don't know.

I also don't want this issue to be ignored because I think that octoprint can do something about it.  There are two ways of thinking about it: Prevention and Mittigation.

- **Prevention**
I had filesystem corruption after power-off on my Linux pc back when ext2 was the norm. Those issues completely went away with ext4 due to the journal used. So a file change is either done completely or not at all. Corruption of files may still be occur if one change is done in several write operations. Then only one of those operations might have succeeded and that causes the corruption. (I still don't understand why "flask" should be written to while octoprint is running.)
I don't understand how the update mechanism is supposed to work and how it safeguards against corrupting the installation in the process. This is usually a complicated issue.

- **Mittigation**
Some parts of Octoprint still seem to be operational. I get the Web site that tells me to check with SSH, I get the lo9g file. So we have the potential to detect this and do something. Again the issue might be caused by something other than octoprint and yes this is not really fixing but more a matter of self defence and covers the real issue up. But if the root cause is a bad SD card then we can do nothing about that and with some action here we might help the user a lot.
What I think of is running something that checks the integrity of octoprint. Maybe delete some "Just in Time" compiled binary stuff, clear caches or whatnot that might be dirty and make sure that all files are available. Maybe even checking checksums and downloading the missing files.
This might be a script simple enough that even I could work on it. If you think that this makes sense I can start doing that. But I would need to know what to check. (I'm sorry that I can not help more)
To prevent covering up the issue this script could also be something that needs to be manually run. So once it works the SSH warn page could be extended to say run this fix install script.

As said before I have only very limited understanding about the inner works of octoprint and if my suggestions makes no sense at all then just ignore me and I apologise for the inconvenience that I caused.

If at least some of it makes sense then I would love to help trying making this situation better. Maybe looking at the corrupted file makes sense? It is a standard octopi installation. Tell me where I can find "flask" and I can take a look or send it to you.

If you have Ideas of what could be done to clean-up then I will test it to see if it works,...

cp2004, 2021-01-17:
All is good, IMO the comment could be interpreted two ways - I went one way, Gina the other. Anyway, we move on.

Re your comments in the 'Prevention' section:
OctoPrint uses `pip` to update it's packages, the standard method of Python packaging. It probably does do some verification, but I don't actually know the specifics. 

To be quite honest, I don't understand why it is the flask section (or others) that are erroring specifically, but what may happen I guess is something in the Python install means that it can no longer find these files. It's not just flask, there's other dependencies affected and even OctoPrint itself - one of the errors is an internal module. And I see you understand filesystems better than I do, so I speak more from experience than knowledge - Raspberry Pi + SD Card + Raspbian has these issues regularly, since the dawn of RPi. I don't really mean *frequently*, but they are always there, coming up every now and then - nothing frequent enough for there to be a real issue in the hundreds of thousands of active installs (>110k, those opted into tracking). 

The mitigation section:
> I get the Web site that tells me to check with SSH

This is the single error page served up by Haproxy from OctoPi, if the server is not around for it to proxy to.

Probably the easiest way to repair a corrupted install, if it is limited to **only** OctoPrint and it's dependencies is to run this command, it sometimes works for people if the Pip install does go wonky:
```
~/oprint/bin/pip install octoprint --force-reinstall --no-cache-dir
```
Which overwrites anything already there with the correct files, also downloading all these files fresh rather than using Pip's cache. I am not *aware* of other simple methods, but again as I said earlier I'm no filesystem expert. 

The rest would probably be standard linux stuff, to repair filesystems etc. if it goes outside OctoPrint's virtualenv. Though I don't know of those either, since it's never something I have had to do personally.

foosel, 2021-01-17:
@JustAnother1 sorry, I really misunderstood you there. Thank you for clarifying and please accept my apologies as well.

JustAnother1, 2021-01-17:
@foosel You are very welcome.

@cp2004 That is really strange. I turned the printer on to check if the "--force-reinstall" works. And it booted. It came up in safe made, due to my configuration in config.yaml. I then wanted to remove that safe mode setting, but I could not find it any more in config.yaml. It also now wanted to upgrade to 1.5.2 So I let it and it worked again. So safe mode safed it???

Regarding the Pi: In the beginning it used ext2 and then these things could easily happen. Especially with people cumming from Windows and not being used to "safely eject" memory and just plugging the Pi out. With ext4 that should not be a big issue any more.

With developers for industrial used electronics the Pi is not loved. Those electronics is more expensive but also more reliable. So people have to now fight against customers that don't understand why they should pay more for a electronic that can do less. So yes the reliability of the PI is not the best it could be. Especially combined with the cheapest sd card and the cheapest USB cabling and power supply there can be issues. Nevertheless it might make sense to still look into those and help fix the issue if possible.

I did not check but I guess that pip will not do much more than checking if a file exists. Extracting files from a zip file is usually safe as that contains a checksum. But pip will probably not check on already existing files. Also if the plugin installed using pip contains a file that is better should not have included that could overwrite something else. (This might be checked by PIP).

The solution I imagined would have been a configuration file that contains a list of all files and their check sums. A script could then check if all files exists and if their check sums match. To calculate the check sum the files have to be read completely. That performance penalty is why I doubt that PIP is doing that. That script could then list the damaged (checksum mismatch) or missing files, or fix the files by downloading the damaged the fixed or missing files. The configuration file with the checksums could be generating when generating the release. But that might be overkill.

If that force reinstall helps then it might be a good Idea to also put it on that HAproxy page?

Due to the effort of reconfiguring and reinstallation of plugins it is probably better to help users fix their installation. My issue seems to be fixed for now. But if the next report comes in maybe do a @JustAnother1 so that I can take a look and try to help the user?

cp2004, 2021-01-17:
That is confusing as to why safe mode fixed it, something is fishy there. I don't know, maybe it is unstable.

The OctoPi haproxy page links to a page on the forums, the force-reinstall option could be added there as a suggestion. I'll try and remember, we usually get those reports on the forums too rather than here. No guarantees, the number of messages I read is *high*.

The script to read/check checksums is a good idea, some sort of scan-and-fix thing. I've done the Python 3 upgrade script, it's not too bad too hard especially when you can auto-detect OctoPi installs.

I'll keep it in mind, if you want to contribute something be sure to look at the community forums or discord server for a more casual chat than a GitHub issue, especially for quick questions. 🙂