OctoPrint/OctoPrint #3962: [Request] Where do I post a request for an entirely new plugin?

KenLucke, 2021-01-16:
I have an idea for an entirely new plugin that I would like to throw out to any developers that would be willing to take it on, but I can't seem to find any place to put such generic request.  Is there such  a place here on the OP github community?
 
 Basically, it would be a  "Translate Position" plugin, that would enable you to shift an already sliced and uploaded gcode file to another location on the bed.  There are various reasons for doing this, and I won't go into them here, but it would be a straightforward x/y translation from the sliced location to a +/- x/y location, so it would be simple math.

jneilliii, 2021-01-17:
The community forum is probably the best place for this at the moment. But we can flag it here as well.

Willmac16, 2021-02-12:
I'd be willing to try to implement this, mostly as a sandbox to work on c/c++ gcode processing and file processing user interface stuff that I will integrate into my more involved gcode processing plugin.

KenLucke, 2021-02-12:
> I'd be willing to try to implement this, mostly as a sandbox to work on c/c++ gcode processing and file processing user interface stuff that I will integrate into my more involved gcode processing plugin.

Let me know when you have something workable to test, and I'll run it through its paces.

Willmac16, 2021-02-19:
@KenLucke the first actually working version should be installable from `https://github.com/Willmac16/OctoPrint-TranslateModel/archive/main.zip`

I still plan to:
* add threading support, so it is non-blocking
* handle already translated objects

I recommend we close this issue and move further discussion to https://github.com/Willmac16/OctoPrint-TranslateModel/discussions/2

closed 2021-02-19 by jneilliii