OctoPrint/OctoPrint #3963: 1.5.2 UnicodeDecodeError on Windows 10 Pro

LAP87, 2021-01-17:
Getting UnicodeDecodeError on Windows 10 Pro, Swedish language/Swe keyboard when trying to serve octoprint.

```
(venv) c:\OctoPrint>octoprint serve
2021-01-17 02:18:03,202 - octoprint.startup - INFO - ******************************************************************************
2021-01-17 02:18:03,203 - octoprint.startup - INFO - Starting OctoPrint 1.5.2
2021-01-17 02:18:03,203 - octoprint.startup - INFO - Starting in SAFE MODE. Third party plugins will be disabled!
2021-01-17 02:18:03,203 - octoprint.startup - INFO - Reason for safe mode: problem during last startup
2021-01-17 02:18:03,203 - octoprint.startup - INFO - ******************************************************************************
2021-01-17 02:18:03,356 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2021-01-17 02:18:03,356 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 8.8.8.8:53 is working
2021-01-17 02:18:03,356 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2021-01-17 02:18:03,378 - octoprint.plugin.core - INFO - Loading plugins from C:\Users\lenny\AppData\Roaming\Python\Python38\site-packages\octoprint\plugins, C:\Users\lenny\AppData\Roaming\OctoPrint\plugins and installed plugin packages...
2021-01-17 02:18:03,604 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin did not pass check, not loading.
2021-01-17 02:18:03,732 - octoprint.plugin.core - INFO - Found 16 plugin(s) providing 16 mixin implementations, 32 hook handlers
2021-01-17 02:18:03,745 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2021-01-17 02:18:03,757 - octoprint.cli.server - ERROR - Uncaught exception
Traceback (most recent call last):
  File "c:\program files (x86)\python38-32\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\program files (x86)\python38-32\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "c:\users\lenny\appdata\roaming\python\python38\scripts\octoprint.exe\__main__.py", line 7, in <module>
    sys.exit(main())
  File "C:\Users\lenny\AppData\Roaming\Python\Python38\site-packages\octoprint\__init__.py", line 939, in main
    octo(args=args, prog_name="octoprint", auto_envvar_prefix="OCTOPRINT")
  File "C:\Users\lenny\AppData\Roaming\Python\Python38\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\lenny\AppData\Roaming\Python\Python38\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "C:\Users\lenny\AppData\Roaming\Python\Python38\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\lenny\AppData\Roaming\Python\Python38\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\lenny\AppData\Roaming\Python\Python38\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "C:\Users\lenny\AppData\Roaming\Python\Python38\site-packages\click\decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "C:\Users\lenny\AppData\Roaming\Python\Python38\site-packages\octoprint\cli\server.py", line 304, in serve_command
    run_server(
  File "C:\Users\lenny\AppData\Roaming\Python\Python38\site-packages\octoprint\cli\server.py", line 160, in run_server
    octoprint_server.run()
  File "C:\Users\lenny\AppData\Roaming\Python\Python38\site-packages\octoprint\server\__init__.py", line 370, in run
    self._start_intermediary_server()
  File "C:\Users\lenny\AppData\Roaming\Python\Python38\site-packages\octoprint\server\__init__.py", line 2376, in _start_intermediary_server
    self._intermediary_server.server_bind()
  File "c:\program files (x86)\python38-32\lib\http\server.py", line 140, in server_bind
    self.server_name = socket.getfqdn(host)
  File "c:\program files (x86)\python38-32\lib\socket.py", line 756, in getfqdn
    hostname, aliases, ipaddrs = gethostbyaddr(name)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xf6 in position 2: invalid start byte

(venv) c:\OctoPrint>
```

jneilliii, 2021-01-17:
@LAP87 you're saying it's not running enough to even create a log file? You can find the log file in the path `%appdata%\OctoPrint\logs` in File Explorer if you're not specifying a base-dir in your command when running.

LAP87, 2021-01-17:
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/5826274/octoprint.log)

LAP87, 2021-01-17:
Added log

LAP87, 2021-01-18:
Temporary/Possible permanent fix for all your Python-related UnicodeDecodeErrors.
Turn on this beta feature in Windows 10 under Region > Additional settings in topright corner > Change date/time > Administrative > Change System Locale.
[image]

This solved the issue for me and i can now continue with setting up Octoprint for windows.

cp2004, 2021-01-18:
Relevant discussion on discord: https://discord.com/channels/704958479194128507/705046975669600298/800826213450383450

foosel, 2021-01-27:
It looks like that this is actually a Python bug for once: 

https://bugs.python.org/issue26227

I take it your computer's host name contains some non ASCII characters, and that's causing this issue. As far as I can see I cannot really do anything about this from my side, so I guess I'll just see that I detect this situation and then link back to this ticket.

LAP87, 2021-01-27:
> It looks like that this is actually a Python bug for once:
>
> https://bugs.python.org/issue26227
>
> I take it your computer's host name contains some non ASCII characters, and that's causing this issue. [...]

Yeah, Frövi-dator is the name of my computer.. doh!
But the UTF-8 beta setting in windows did solve the issue, but good to know! I was always wondering why some things worked on my other older computer vs my gaming rig.

foosel, 2021-01-27:
1.6.0 will ship with a more meaningful error message:

```
2021-01-27 17:28:37,760 - octoprint.server - DEBUG - Starting intermediary server on http://0.0.0.0:5000 and http://[::]:5000
2021-01-27 17:28:37,761 - octoprint.startup - CRITICAL - OctoPrint cannot start due to a Python bug (https://bugs.python.org/issue26227). Your computer's host name contains non-ASCII characters. Please either change your computer's host name to contain only ASCII characters, or take a look at https://github.com/OctoPrint/OctoPrint/issues/3963 for other options.
2021-01-27 17:28:37,761 - octoprint.startup - CRITICAL - There was a fatal error starting up OctoPrint.
```

commit eff9793b (2021-01-27): 🗑 Better error handling for a python bug

Non-ASCII host names under Windows are bad news for firing up an
HTTPServer instance. We can't fix that, but we can display a better
error message.

See https://bugs.python.org/issue26227

Closes #3963
closed 2021-04-27 by foosel