OctoPrint/OctoPrint #3965: Printer NOT OFF. M81.

Foxcor, 2021-01-17:
#### What were you doing?

GCODE after printing
```
M104 S0 ; Turn off Extruder temperature
M140 S0 ; Turn off Heat Bed
M84 ; Disable stepper motors
G4 S1; waits 1 second
M109 R50 ; Wait for extruder cooldown
M150 R255
M81 ; Turn OFF printer
```

#### What did you expect to happen?
Printer OFF
#### What happened instead?
Printer not OFF
#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint
Octoprint 1.5.2

#### Operating System running OctoPrint

OctoPi 0.18.0 RC2 x64
```
browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.141 Safari/537.36
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 3925356544
env.os.bits: 64
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.2.4
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.2

```
#### Printer model & used firmware incl. version

BTT SKR 1.4 Turbo, Marlin 2.0.7.2
#### Browser and version of browser, operating system running browser

#### Link to octoprint.log
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/5825840/octoprint.log)

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

Printer not OFF. 
LOG
```
Recv: echo:busy: processing
Recv: //action:notification Foxcor's Mega S V261220 Готов.
Recv: ok
Send: N44615 M150 R255*101
Recv: ok
Send: N44616 M81*27
Recv: M106 P0 S0
Recv: //action:notification Foxcor's Mega S V261220 OFF.
Recv: ok
Recv:  T:50.74 /0.00 B:68.66 /0.00 @:0 B@:0
Recv: ok
Changing monitoring state from "Finishing" to "Operational"
```
But if you send the command M81 directly, through the WEB, through the screen, through plugins, everything works and turns off.
```
Send: M81
Recv: M106 P0 S0
Recv: //action:notification Foxcor's Mega S V261220 OFF.
Recv: ok
Recv:  T:47.88 /0.00 B:62.04 /0.00 @:0 B@:0
Recv:  T:47.88 /0.00 B:62.04 /0.00 @:0 B@:0
Recv:  T:47.81 /0.00 B:61.99 /0.00 @:0 B@:0
Recv:  T:47.88 /0.00 B:61.90 /0.00 @:0 B@:0
Recv:  T:47.81 /0.00 B:61.85 /0.00 @:0 B@:0
Recv:  T:47.81 /0.00 B:61.85 /0.00 @:0 B@:0
Recv:  T:47.81 /0.00 B:61.81 /0.00 @:0 B@:0
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:3831
Please see https://faq.octoprint.org/serialerror for possible reasons of this.
Changing monitoring state from "Operational" to "Offline (Error: SerialException: 'read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:3831)"
Connection closed, closing down monitor
```
The problem is with the shutdown by GCODE Scripts, after printing.

cp2004, 2021-01-17:
This is not an OctoPrint issue, since the command is sent and received fine by the firmware - you can see the reply:
> ```
>    Send: M81
>    Recv: M106 P0 S0
>    Recv: //action:notification Foxcor's Mega S V261220 OFF.
>    Recv: ok
>    ```

Maybe there is some sort of protection in the printer firmware, that doesn't shut down the printer when it thinks it is printing?

There's nothing else OctoPrint can do here, if the command is sent and acknowledged then it is up to your printer firmware to sort out the rest.

jneilliii, 2021-01-17:
Or this could be a plugin issue, it seems you have PSU Control plugin installed and that is what is probably controlling your M81 commands to power off a relay to your PSU? If so then the issue is on the plugin side.

Foxcor, 2021-01-17:
@cp2004 
> Maybe there is some sort of protection in the printer firmware, that doesn't shut down the printer when it thinks it is printing?

Maybe, but....
If I send M81 from Octoprint terminal, the printer turns off.
If I use shutdown via PSU Control, the printer turns off.
If I press the shutdown button on the printer screen, the printer turns off.
In Marlin, PSU control is just a few lines away:
```
/**
 * Power Supply Control
 *
 * Enable and connect the power supply to the PS_ON_PIN.
 * Specify whether the power supply is active HIGH or active LOW.
 */
#define PSU_CONTROL
#define PSU_NAME "Power Supply"

#if ENABLED(PSU_CONTROL)
  #define PSU_ACTIVE_STATE HIGH      // Set 'LOW' for ATX, 'HIGH' for X-Box

  //#define PSU_DEFAULT_OFF         // Keep power off until enabled directly with M80
  //#define PSU_POWERUP_DELAY 250   // (ms) Delay for the PSU to warm up to full power

  //#define PSU_POWERUP_GCODE  "M355 S1"  // G-code to run after power-on (e.g., case light on)
  //#define PSU_POWEROFF_GCODE "M355 S0"  // G-code to run before power-off (e.g., case light off)

  #define AUTO_POWER_CONTROL      // Enable automatic control of the PS_ON pin
  #if ENABLED(AUTO_POWER_CONTROL)
    #define AUTO_POWER_FANS         // Turn on PSU if fans need power
    #define AUTO_POWER_E_FANS
    //#define AUTO_POWER_CONTROLLERFAN
    //#define AUTO_POWER_CHAMBER_FAN
    #define AUTO_POWER_E_TEMP        50 // (°C) Turn on PSU if any extruder is over this temperature
    //#define AUTO_POWER_CHAMBER_TEMP  30 // (°C) Turn on PSU if the chamber is over this temperature
    #define POWER_TIMEOUT              3600 // (s) Turn off power if the machine is idle for this duration
    #define POWER_OFF_DELAY          5 // (s) Delay of poweroff after M81 command. Useful to let fans run for extra time.
  #endif
#endif
```
Since the shutdown command in the final GCODE comes with the print number
`"Send: N44616 M81 * 27"`
those. the printer thinks it is still printing and perhaps Marlin is protecting somehow the execution of the M81 while it is printing.
In my humble opinion, GCODE Scripts should not be executed within the main code in the file. At least such a concept is on my screen from BTT TFT35 V3.0.

cp2004, 2021-01-17:
Line numbers and checksums are used for all communication, through the terminal in OctoPrint, or scripts, or the gcode file itself. That won't make a difference.

Ok let's look at it this way - what can OctoPrint do differently to make this work?

The command is even acknowledged by your printer. It replies back saying it will turn off, if it doesn't then what does OctoPrint do?
 
Open an issue or ask someone from the Marlin discord - it's not an OctoPrint issue.

cp2004, 2021-01-17:
Also, a full serial log file would be needed if you still think there is something broken with OctoPrint, since we cannot tell the difference with just a short snippet.

Foxcor, 2021-01-17:
@cp2004 

> Also, a full serial log file would be needed if you still think there is something broken with OctoPrint, since we cannot tell the difference with just a short snippet.

Ok

Foxcor, 2021-01-17:
@cp2004 
I don't understand what happened, but when serial.log is enabled, the printer behaves inappropriately. Bugs, wrong commands, etc. come up. I can't make a full log :(

foosel, 2021-01-25:
We cannot help you without a log of said misbehaving.

cp2004, 2021-02-20:
Hey @Foxcor

Since we couldn't identify an issue with OctoPrint and the additional information requested was not provided, I'm going to close this. Feel free to request a reopen if you find further issues.

closed 2021-02-20 by cp2004