OctoPrint/OctoPrint #3968: Serial desconnects just as the print begins

jj-uk, 2021-01-17:
#### What were you doing?
Sent a print to OctoPrint. The printer wipes the nozzle, then moves to the start position, Z axis is lowered to the start position, then the printer serial disconnects.

This has happened many times now; it always disconnects just as the print is about to begin, at the same point each time; and on different print jobs of different items.

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

#### What did you expect to happen?
Print as normal.

#### What happened instead?
Lost serial connection.

#### Did the same happen when running OctoPrint in safe mode?
I've not tried; I restart the printer and octoprint, and it worked as normal.

#### Version of OctoPrint
1.5.2

#### Operating System running OctoPrint
OctoPrint 1.5.2 Python 2.7.13 OctoPi 0.16.0

#### Printer model & used firmware incl. version
2019 Anycubic i3 Mega (not mega S). Firmware v1.1.2

#### Browser and version of browser, operating system running browser
Chrome / Win10

#### Link to octoprint.log

https://gist.github.com/jj-uk/0894cffd6e9c15781181e2a36366c665

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

jneilliii, 2021-01-17:
This doesn't seem to be a bug with OctoPrint based on your log. Your printer stops talking/responding to OctoPrint so therefore OctoPrint kills the connection because the communication is not there anymore. You should definitely try to see if you can reproduce this running OctoPrint in safe mode, to rule out any potential issues caused by plugins that may be related, but in your instance I think this is more of a connection issue. Check your cables, try a different USB cable, different USB port on your Pi, etc. Some people have even gone as far as grounding the USB housing of the Pi to their printer's shared ground from PSU to avoid serial noise/electromagnetic interference. Also if your printer gets power from your pi over the USB connection when the printer is powered off, that is something that has been reported to cause this. If that is the case, this link may be relevant to you.

https://community.octoprint.org/t/put-tape-on-the-5v-pin-why-and-how/13574

I highly recommend you check the forum for additional help here, as it doesn't appear to be a bug with OctoPrint.

closed 2021-01-17 by jneilliii

jj-uk, 2021-01-18:
Thanks for the help. I'm really struggling to connect since updating to 1.5.2. I have to click the connect button multiple times to get it to connect after restarting.

I've started the server in safe-mode and it connects just fine, first time, every time.  So, I guess its a plugin causing the issue?

jj-uk, 2021-01-18:
Update: I've disabled my plugins one-by-one, and it appears to be "Cancel Object v0.4.4" that's causing the issue.

https://github.com/paukstelis/OctoPrint-Cancelobject 

I'll open an issue over there. Thanks for your help.