OctoPrint/OctoPrint #3969: [Request] Ability to suppress the "blocked command" popup

fluffy-critter, 2021-01-17:
**Is your feature request related to a problem? Please describe.**

One of the slicers I use (Canvas3D) sends M0 as its initial command, which is blocked. There doesn't seem to be any way to get the slicer to not send this command, but every time I load a gcode file with that I get another little popup message in the corner of my Octoprint UI, which gets in the way of things and is kind of annoying to clear due to the small click target.

**Describe the solution you'd like**

I'd like to be able to have those "suppressed command" popups have configuration for how long they stick around, with the ability to turn them off entirely if so desired.

cp2004, 2021-01-17:
This is already an option! So you are in luck.

Here's a screenshot, find it under Settings > Serial connection > Behaviour tab, scroll down a little to find it.

[image]

closed 2021-01-17 by cp2004

fluffy-critter, 2021-01-17:
Ah, thanks! I had looked around for it all over but I never thought to look in the "serial connection" tab.