OctoPrint/OctoPrint #3971: [Request] Easy plugin manager search dialogs ergonomy enhancement

yet-another-average-joe, 2021-01-19:
**Is your feature request related to a problem? Please describe.**

Not really a problem. An ergonomy enhancement...
Goto the Plugin Manager. When searching for a plugin, first thing that comes in mind is typing a keyword or a name in the search field, despite it is entitled "Installed Plugins" ! We obviously get an empty list and now the "Get More..." button becomes more visible. We click that button... And we have to type the keywords again in the "Install new Plugins..." dialog.

**Describe the solution you'd like**

The "Install new Plugins..." search field value could be copied from the parent dialog. Would save some keystrokes, and be more intuitive.

cp2004, 2021-01-19:
With the changes in #3835 (for 1.6.0) I don't think this will be as relevant, since the get more button has moved (see the PR for a gif of the updated UI).

It would be hard to differentiate between a search term that *shouldn't* be moved to the plugin repository search bar. Since now you have the opposite problem, of annoying someone who has to delete a search term they didn't put there.

Personally not for this suggestion, especially with the get more button moving IMO this is less likely be a problem. Look forward to someone else's opinion.

foosel, 2021-01-25:
I gotta say, I agree with @cp2004 here. 

For reference, here's how it'll look like in 1.6+:

[image]

IMHO it's fairly clear that the search field under "Installed Plugins" only relates to those, and carrying over search terms from that into the search bar in the plugin repository viewer would create more issues than solve I fear. 

What I'm wondering though is if it could maybe be made possible to have both search bars share an "autocomplete from history" thing... Thoughts on that?