OctoPrint/OctoPrint #3972: bugfix/enhancement timelapse rendering with ffmpeg killed (memory on raspi, improved -g 5 flag helps)

ozett, 2021-01-20:
#### What did you expect to happen?
render the timlapse image files witout error.
images exist in /tmp

#### What happened instead?
error 9 from ffmpeg. 
(investigated with cmd-line execution that it is killed while rendering the images to a mp4 on raspberry)

#### Did the same happen when running OctoPrint in safe mode?
not tried. solution already found: add flag -g 5 to ffmpeg 

If you can't test this in safe mode, state why.
-->solution already found: add flag -g 5 to ffmpeg

#### Version of OctoPrint

#### Operating System running OctoPrint
OctoPrint 1.5.3 Python 3.7.3 OctoPi 0.18.0

----
error: 
while rendering of the images which octopi took with a rtsp-cam in h264 and 3120x2400

Snapshot-Url:
```http//192.168.14.115/Streaming/channels/102/picture```

 killed silently the ffmpeg process.

i recorded the cmd-line from octopi rendering with ffmpeg and did it by hand to test some flags.
i found that adding "-g 5" solved the problem

my succesful test-command, i dont want [to patch the timelapse.py like here](https://github.com/OctoPrint/OctoPrint/issues/2373)
```
/usr/bin/ffmpeg -s 1920x1080 -loglevel error -framerate 25 -fflags discardcorrupt -i /home/pi/.octoprint/timelapse/tmp/slug6_deformed_repaired_0.2mm_PLA_MINI_33m_20210120091941-%d.jpg -vcodec libx264 -threads 2 -crf 35 -g 5 -y -f mp4 /home/pi/.octoprint/timelapse/slug8.mp4
```
works. 

without ```-g 5``` the ffmpeg-process silently dies...

also maybe worth to check if all images are not corrupted before start rendering
```-fflags discardcorrupt```

may you want to add this to the ffmpeg options in octopi-timelapse.py ?
([was already suggested as improvement here](https://github.com/OctoPrint/OctoPrint/issues/2188))

ozett, 2021-01-20:
thanks bot for approval. hard to fill in the formula, but i want to help

foosel, 2021-01-20:
As far as I understand the `-g` option, it sets the group of pictures size, as in, how many pictures per intra frame. Can you explain *why* that solved your issue here? I don't see how this could prevent any kind of silent crashing issues and this sounds rather like there might be a debug flag missing here that would make encoding fail more visibly.

ozett, 2021-01-21:
i had the impression that ffmpeg dies from insufficient memory while rendering my 300 pictures of 3120x2600 something.

so i searched the internet and found that the -g oprtion should helps with something of interframes, that occures to me as making smaller groups of renderimages in memory. somehow. i will look the hint up and put it here.

it worked as i tested it, so i found it was the solutiion.
with the normal command the rendering was killed always after 40-44 frames.
i before tried to limit the stock octo-timelapse command for rendering only 30 frames. that worked.
thats why i assumed some memory issues for rendering my rtsp-snapshots.

ozett, 2021-01-21:
made a test again: failed with stock octo-timelapse:

[image]

captured command-line from ps:
```
 /usr/bin/ffmpeg -framerate 25 -i /home/pi/.octoprint/timelapse/tmp/slug6_deformed_repaired_0.2mm_PLA_MINI_1h10m_20210120140409-%d.jpg -vcodec libx264 -threads 2 -r 25 -y -b 10000k -f mp4 -vf [in] format=yuv420p [out] /home/pi/.octoprint/timelapse/.slug6_deformed_repaired_0.2mm_PLA_MINI_1h10m_20210120140409.mp4
```
(somehow too much ) error-log from the error-box:
```
ffmpeg version 4.1.6-1~deb10u1+rpt1 Copyright (c) 2000-2020 the FFmpeg developers
,  built with gcc 8 (Raspbian 8.3.0-6+rpi1)
,  configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared --libdir=/usr/lib/arm-linux-gnueabihf --cpu=arm1176jzf-s --arch=arm
,  WARNING: library configuration mismatch
,  avutil      configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a7 --arch=armv6t2 --disable-thumb --enable-shared --disable-doc --disable-programs
,  avcodec     configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a7 --arch=armv6t2 --disable-thumb --enable-shared --disable-doc --disable-programs
,  avformat    configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --ena [... cut ...]
,frame=   37 fps=2.5 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   0x    
,frame=   39 fps=2.5 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   0x    
,frame=   41 fps=2.5 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   0x    
,frame=   43 fps=2.5 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   0x    
,frame=   44 fps=2.3 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   0x,    
,frame=   45 fps=1.2 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   0x    
```

ozett, 2021-01-21:
only adding ```-g```  and success... 

```
/usr/bin/ffmpeg -framerate 25 -i /home/pi/.octoprint/timelapse/tmp/slug6_deformed_repaired_0.2mm_PLA_MINI_1h10m_20210120140409-%d.jpg -g 5 -vcodec libx264 -threads 2 -r 25 -y -b 10000k -f mp4 /home/pi/.octoprint/timelapse/slugtest20210121-1.mp4`
```

[image]

foosel, 2021-01-25:
Since with #3977 there just has been another user who ran into a problem that looks identical, I'll look into this (and potentially some parameter editing option too).

ozett, 2021-01-25:
Yeah, thanks for looking into this "-9" bug. 😄 

would love to see some option-input for stock octo-timelapse ffmpeg parameters. so one could easily make fade-in and fade-out.

small examples: 
https://medium.com/@sekhar.rahul/creating-a-time-lapse-video-on-the-command-line-with-ffmpeg-1a7566caf877

Than people could have the idea to share such ffmpeg snippes 👍

foosel, 2021-03-24:
1.6.0 will allow the full configuration of the whole command line, so you can tweak and tune things as needed. Hidden under advanced options however, as you can also break things that way.

[image]

ozett, 2021-03-24:
Great. 
you must have felt that I was yesterday wondering if there was a patch coming up...  i will try
👍

ozett, 2021-03-24:
[image]

😭 😭 😭 😭 😭  and i did not find any dev 1.6 available to test on my octo-pi 😭 😭 😭 😭 😭

cp2004, 2021-03-24:
This is just a small part of what goes into a release, switch to the 'Maintenance RCs' release channel to be notified when a release candidate becomes available.

foosel, 2021-03-24:
And realize that I do this project still primarily all on my own, am struggling with way more than just this single request and have to somehow keep all balls in the air at the same time. If stuff is too slow, consider contributing.

ozett, 2021-03-24:
you do great! 

(and i wish i could code at all ...)
sorry, if any misunderstandings. this all is a great work i will wait patiently (and now silently) .... 
[image]

ozett, 2021-03-27:
the invocing command line for ffmpeg is not in the logs. (?)

there is only the long error-report from ffmpeg, but not the resulting cmd-line from octoprint, 
maybe it should be logged for further analysis to see what causes an error.

(or i dont find the right place, i am looking in octoprint.log....)
should i make another issue or could it be embedded with this issue in 1.6?

cp2004, 2021-03-27:
That would make sense, that if it failed OctoPrint logs the configured command so we can sort out people who put a typo in or something.

ozett, 2021-03-27:
i first dont wanted to mention it here to not extend this, but if improvements go into 1.6 maybe
it would be great to have a "reset-to-default"-Button for the pre-configured cmd-line. 
in case anybody messed all commands totally up.

one could also think about saving differnt cmd-lines as some profiles (can be a new feature ?)
ie. with different rendering settings, or from different experiments with commands...

ozett, 2021-03-27:
the option -b should be used as -b:v or -b:a (video/audio),
but maybe for the resulting command-line it should not be empty..

fiddling with my error i found that this option-value seems also depend on the encoder, too:
[image]

foosel, 2021-03-29:
1.6.0 is in feature freeze already, I'm currently busy making everything ready for the first RC which I hope to push out tomorrow. I wasn't aware of this (changed?) behaviour of the `b` parameter, so this sounds also like it might be something that requires some extensive testing and a UI overhaul to solve, so definitely not something I feel comfortable shoe-horning into 1.6 still. Have extracted a separate issue with a milestone of 1.7.0.

foosel, 2021-04-27:
1.6.0 has just been released.

closed 2021-04-27 by foosel