OctoPrint/OctoPrint #3973: Gpio pins not working

Octoprintz, 2021-01-20:
#### What were you doing?

1. Instaled latest version of Octoprint
2. Found that Octolight was released
3. Instaled Octolight
4. Uninstalled enclosure plugin

#### What did you expect to happen?
GPIO Pin 10 to turn on

#### What happened instead?
Did not turn on, did multiple tests with voltimeeter using two different plugins
#### Did the same happen when running OctoPrint in safe mode?
No because the plugins where disabled and I was not able to test it.

#### Version of OctoPrint
1.5.3

#### Operating System running OctoPrint
Octopi 0.18.0

#### Printer model & used firmware incl. version
Problem with external lights not printer

#### Browser and version of browser, operating system running browser
Chrome

#### Link to octoprint.log
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/5844682/octoprint.log)

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

jneilliii, 2021-01-20:
This is a plugin issue, not an OctoPrint bug. For help with plugin you should reach out to the respective plugin's issue trackers directly or ask for help in the community forum or Discord.

closed 2021-01-20 by jneilliii

Octoprintz, 2021-01-20:
No, this problem occurred with two different gpio controlling plugins. One of the plugins I was already using and it worked but after I installed the newest version of Octoprint it did not work.

jneilliii, 2021-01-20:
OctoPrint has nothing to do with GPIO so definitely sounds like a compatibility bug with those plugins.

cp2004, 2021-01-20:
It is *still* not an OctoPrint issue, since OctoPrint doesn't even know GPIOs exist. It has no code within it that would deal with GPIO, it will run on devices that don't use GPIO.

If something is broken with GPIO then report it to the plugin.