OctoPrint/OctoPrint #3974: Marlin binary file transfer [Request]

tiomel, 2021-01-20:
It would be relatively easy to introduce an adjustment in the options so that octoprint uses the G29 B1 command instead of the G29 in order to make use of the Marlin Binary File transfer function and make copies to the microsd faster when this option is duly activated in Marlin .

There is a plugin but it hasn't worked for me at least.

https://marlinfw.org/docs/gcode/M028.html

foosel, 2021-01-21:
Please don't start a feature request with "it would be easy" if you are not the person who is offering to do and maintain it ;) A request like this can have severe repercussions for workflows differing from yours (in this case I can't just blindly send a parameter for a command unless I have verified the firmware knows what to do with it, or I might break things for everyone whose firmware *doesn't* support it), and looking at the available documentation and the plugin you mentioned it's also not a case of "just" adding a parameter to a command.

Speaking of plugins, as there already is a plugin that supports this but for some reason doesn't work for you, then please raise an issue with the plugin's author and work with them to resolve this. It doesn't make sense for me to do competing implementations of functionality and the goal in general is to *extract* functionality from core OctoPrint into plugins instead of merging it in.

closed 2021-01-21 by foosel

tiomel, 2021-01-21:
I had only proposed a feature that Octoprint lacks and that many people are looking for. I propose that it be introduced as an optional option for those of us who have this function activated in the firmware. Many people are frustrated with how files are copied to SD.

Isn't this a requirement for a new feature?

Thanks and greetings, I'm sorry to have upset you so much.

cp2004, 2021-01-21:
It's available in a plugin already, if it were me with this plugin I'd be quite upset someone had steam rolled over my existing implementation, that I had worked hard on. There's no reason for us to duplicate this and put it into the base OctoPrint install.
Binary file transfer (and the OctoPrint plugin) did work for me when tried it out, so if something doesn't work for you then it can probably be fixed by the author, or a specific problem with your general setup.

foosel, 2021-01-21:
@tiomel you haven't upset me, I just wanted to point out that it's easy to say something is "easy to implement" when in fact it isn't. As pointed out by now both me and @cp2004, reporting your problem to the plugin author makes more sense than reimplementing a perfectly good example of the ecosystem and community working like it should and potentially alienating the plugin author in the process.

tiomel, 2021-01-21:
The installation of this plugin fails in any modern installation of octoprint, and it is acknowledged by the author who claims he does not have time to correct the problem.
On the other hand, the plugin as I have seen requires that you manually upload each file with a complicated routine that is not useful, it would be more convenient to be able to activate this by default so that simply when you copy any file to the SD it does so in an invisible and automated way .
I think this would be a very good feature for many people who want to make long prints from SD and have the printer away from the computer.
If you're not interested, I'll work on this arrangement myself so that I can use this feature that the people at Marlin have developed for us.

foosel, 2021-01-21:
Why didn't you say so in the first place? Based on the information you so far shared, it came across as "the plugin didn't work for me, so now I'm asking you to implement this", not "I already looked into this and the plugin is no longer actively maintained".

In that case I'd strongly suggest to report the plugin as abandoned on the plugin repository so it can hopefully be adopted by someone interested in maintaining this feature, and maybe also adjusted so the normal file upload process can be overwritten instead of utilizing its own mechanism. 

The goal of OctoPrint's core implementation (which you are currently looking at) is not to support every possible feature and workflow under the sun. That's how you get a code base that can no longer be maintained (especially not by primarily only one person, hi 👋!). The plugin system allows to spread the burden of maintenance across the shoulders of many, with OctoPrint aiming instead to make extending it in whatever direction easier. It's not that I don't want a feature like this to work with OctoPrint or am "not interested" in the improvements it might bring, I really do. It's that I would rather see the community that's interested into this to step up, implement and maintain it, before I have yet another bunch of code in core to take care of that further complicates an already complicated communication layer. Please understand the reasoning behind this.

tiomel, 2021-01-21:
Ok, thank you very much for your time, I just thought it would be simple and interesting to implement.
Regards.

jneilliii, 2021-01-21:
I'd like to point out that OctoPrint is not a Marlin only product as well. Implementing something in core that is firmware specific, and optionally enabled within the firmware (not enabled by default I don't think) that does support it would make way more sense as a plugin. Also, from information I know from the OctoFarm discord and the major contributor of the binary file transfer stuff to Marlin, it's still being changed, and therefore makes even more sense to be maintained as a plugin, because that can be adapted a lot quicker than the core OctoPrint can.