OctoPrint/OctoPrint #3975: No such file or directory: 'sudo' - Restarting OctoPrint

Lithimlin, 2021-01-21:
#### What were you doing?

1. Have OpctoPrint installed via the docker container
2. Try to restart OctoPrint either manually or because something was updated
3. OctoPrint won't restart but instead print errors in the log

#### What did you expect to happen?

OctoPrint restarts

#### What happened instead?

When restarting due to an update, the following error occurs:
```
2021-01-21 10:25:01,964 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2021-01-21 10:25:01,965 - octoprint.plugins.softwareupdate - INFO - Restarting...
Exception in thread Thread-31:
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/sarge/__init__.py", line 655, in run
    self.process = p = Popen(self.args, **self.kwargs)
  File "/usr/local/lib/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/local/lib/python3.8/subprocess.py", line 1702, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'sudo'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.8/site-packages/sarge/__init__.py", line 1187, in run_node
    result = getattr(self, method)(node, input, async_)
  File "/usr/local/lib/python3.8/site-packages/sarge/__init__.py", line 1333, in run_command_node
    node.cmd.run(input=input, async_=async_)
  File "/usr/local/lib/python3.8/site-packages/sarge/__init__.py", line 658, in run
    raise ValueError('Command not found: %s' % self.args[0])
ValueError: Command not found: sudo
```

When trying to restart manually, it is the following error:

```
2021-01-21 10:28:27,047 - octoprint.server.api.system - WARNING - Command for core:restart failed with return code 127:,
! STDERR: /bin/sh: 1: sudo: not found,
! STDOUT: ,
2021-01-21 10:28:27,024 - octoprint.server.api.system - INFO - Performing command for core:restart: sudo service octoprint restart
```

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint
1.5.3

#### Operating System running OctoPrint
Linux

#### Printer model & used firmware incl. version
--

#### Browser and version of browser, operating system running browser
Firefox 84.0.2 (64-Bit)

#### Link to octoprint.log
see above

#### Link to contents of terminal tab or serial.log
--

#### Link to contents of Javascript console in the browser
--

#### Screenshot(s)/video(s) showing the problem:
--

I have read the FAQ.

foosel, 2021-01-21:
That sounds like a misconfiguration rather than a bug. Please get in touch on the [community forums](https://community.octoprint.org) or the [discord server](https://discord.octoprint.org).

closed 2021-01-21 by foosel

cp2004, 2021-01-21:
There's documentation on the correct command to restart OctoPrint on the [official docker container repository.](https://github.com/OctoPrint/octoprint-docker#restarting-octoprint)