OctoPrint/OctoPrint #3976: TRouble upgrading from OctoPrint: 1.4.0 to 1.5.3

wookie73, 2021-01-21:
Trying to update from 1.4 to 1.5.3
Update fails.
I have read the release notes and don't have any of the incompatible plug ins.
All installed plugins are up to date.

Error in octoprint.log is:
2021-01-21 14:55:17,327 - octoprint.plugins.softwareupdate - INFO - Starting update of octoprint to 1.5.3...
2021-01-21 14:55:35,859 - octoprint.plugins.softwareupdate - ERROR - Update of octoprint can not be performed, please also check plugin_softwareupdate_console.log for possible causes of this
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 1060, in _perform_update
    update_result = updater.perform_update(target, populated_check, target_version, log_cb=self._log, online=online)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/updaters/pip.py", line 116, in perform_update
    raise exceptions.UpdateError("Error while executing pip install", (stdout, stderr))
UpdateError
2021-01-21 14:55:36,083 - octoprint.plugins.tracking - INFO - Sent tracking event update_failed, payload: {'to_version': '1.5.3', 'from_version': '1.4.0', 'target': u'octoprint'}

Error in plugin_softwareupdate_console.log.txt is:
-01-21 14:55:35,842 > Installing build dependencies: finished with status 'error'
2021-01-21 14:55:35,847 ! ERROR: Command errored out with exit status 1:
2021-01-21 14:55:35,848 ! command: /home/pi/oprint/bin/python2 /home/pi/oprint/local/lib/python2.7/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-f8bGvZ/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- setuptools wheel Cython
2021-01-21 14:55:35,848 ! cwd: None
2021-01-21 14:55:35,850 ! Complete output (24 lines):
2021-01-21 14:55:35,850 ! Traceback (most recent call last):
2021-01-21 14:55:35,850 ! File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
2021-01-21 14:55:35,851 ! "__main__", fname, loader, pkg_name)
2021-01-21 14:55:35,851 ! File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
2021-01-21 14:55:35,852 ! exec code in run_globals
2021-01-21 14:55:35,852 ! File "/home/pi/oprint/lib/python2.7/site-packages/pip/__main__.py", line 16, in <module>
2021-01-21 14:55:35,852 ! from pip._internal.main import main as _main  # isort:skip # noqa
2021-01-21 14:55:35,852 ! File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/main.py", line 13, in <module>
2021-01-21 14:55:35,853 ! from pip._internal.cli.autocompletion import autocomplete
2021-01-21 14:55:35,853 ! File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/cli/autocompletion.py", line 11, in <module>
2021-01-21 14:55:35,854 ! from pip._internal.cli.main_parser import create_main_parser
2021-01-21 14:55:35,854 ! File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/cli/main_parser.py", line 7, in <module>
2021-01-21 14:55:35,854 ! from pip._internal.cli import cmdoptions
2021-01-21 14:55:35,854 ! File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/cli/cmdoptions.py", line 25, in <module>
2021-01-21 14:55:35,855 ! from pip._internal.locations import USER_CACHE_DIR, get_src_prefix
2021-01-21 14:55:35,855 ! File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/locations.py", line 18, in <module>
2021-01-21 14:55:35,856 ! from pip._internal.utils import appdirs
2021-01-21 14:55:35,856 ! File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/utils/appdirs.py", line 16, in <module>
2021-01-21 14:55:35,856 ! from pip._internal.utils.compat import WINDOWS, expanduser
2021-01-21 14:55:35,856 ! File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/utils/compat.py", line 17, in <module>
2021-01-21 14:55:35,857 ! from pip._vendor.urllib3.util import IS_PYOPENSSL
2021-01-21 14:55:35,857 ! File "/home/pi/oprint/lib/python2.7/site-packages/pip/_vendor/urllib3/__init__.py", line 21, in <module>
2021-01-21 14:55:35,857 ! from logging import NullHandler
2021-01-21 14:55:35,858 ! ImportError: cannot import name NullHandler
2021-01-21 14:55:35,858 ! ----------------------------------------
2021-01-21 14:55:35,858 ! ERROR: Command errored out with exit status 1: /home/pi/oprint/bin/python2 /home/pi/oprint/local/lib/python2.7/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-f8bGvZ/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- setuptools wheel Cython Check the logs for full command output.

Any pointers on what to do to get update sorted much appreciated.

Thanks Dave

Drummerboy2, 2021-01-21:
[octoprint (4).log](https://github.com/OctoPrint/OctoPrint/files/5851186/octoprint.4.log)
I get the message "System update not started. Is it running? check log"

wookie73, 2021-01-22:
Hi
Sorry should have posted it on the forum my mistake.
Just for reference was running an older version of Python 2.x. Upgraded to Python 3.x which seem to fix the issue.

closed 2021-01-22 by wookie73