OctoPrint/OctoPrint #3984: [Request] Add a hook to allow plugins to prevent a connect from going through

foosel, 2021-01-25:
**Is your feature request related to a problem? Please describe.**

This came up in the FirmwareUpdater plugin's repository:

https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/132

tldr: If a firmware update is in progress, a connect triggered by the user (or some other API client) will break the flash or verify.

**Describe the solution you'd like**

Plugins should be allowed to subscribe to a hook that would be queried by the printer class upon connection initialization and allow plugins to return some value (`False` or maybe an error message, or maybe even raise an exception?) to prevent the connect from going through.

**Describe alternatives you've considered**

Plugins monkey patching `self._printer.connect` to prevent connections from going through. Dirty, ugly, not recommended.

**Additional context**

We should also think about other places hooks like that could make sense.

foosel, 2021-03-24:
Ready for 1.6.0

benlye, 2021-04-05:
Just updated the Firmware Updater plugin to use this hook and tested on 1.6.0rc1 - works great, thanks!

commit 43e5c654 (2021-03-24): ✨ Add hook "octoprint.printer.handle_connect"

Can be used to intercept connect command on Printer interface and
prevent or just hook into.

Closes #3984
closed 2021-04-27 by foosel