OctoPrint/OctoPrint #3985: [Request] Error reporting for broken GCODE scripts

foosel, 2021-01-25:
**Is your feature request related to a problem? Please describe.**

I have just had an extremely annoying situation where my prints repeatedly failed because I had a plugin referenced in the gcode "before print job" section that I'd uninstalled, and my printer was no longer bed levelling because the gcode script wasn't executing. It took me ages to figure this out.

Errors in custom gcode really need to be flagged up somewhere - personally I'd expect the job to fail and a dialog to pop up. Could things like broken templates be flagged up at the time you attempt to save them in settings?

_Originally posted by @Jonty in https://github.com/OctoPrint/OctoPrint/issues/2414#issuecomment-734491009_

**Describe the solution you'd like**

Try to prerender gcode scripts with an example context upon save and check if an error gets produced, if so refuse to save and show a warning to the user. Maybe a testing endpoint is needed that can be called from the frontend?

**Describe alternatives you've considered**

None

**Additional context**

#2414

kantlivelong, 2021-02-15:
> Try to prerender gcode scripts with an example context upon save and check if an error gets produced, if so refuse to save and show a warning to the user.

Just need to keep in mind how that may affect plugins that use `octoprint.comm.protocol.scripts` such as [OctoPrint-SmartPreHeat](https://github.com/kantlivelong/OctoPrint-SmartPreheat/blob/0896e8101da52cbb792aabf6bcd768cde1d64ddd/octoprint_smartpreheat/__init__.py#L112-L122). Other plugins out there could be adding GPIO to the mix as well.

foosel, 2021-02-15:
Shouldn't that be covered by rendering the actual script though as long as the script hooks are called?

kantlivelong, 2021-02-15:
I think that would depend on the implementation of the plugin since as it stands plugins are expecting these hooks to be run during a job rather than a dryrun. In the case of the plugin above simulating a job via a dryrun would work out okay. I'm not sure if there are any plugins out there that might interface with hardware out there that utilize these hooks and depend on real world interaction. Likely low risk issue but just throwing the possibility out there.

Jonty, 2021-02-15:
It would probably make sense to make it mandatory for plugins to be able to support this use case, which I assume would be a breaking change in the API?