OctoPrint/OctoPrint #3987: You haven't yet decided on whether to enable or disable the online connectivity check. You need to either enable or disable it before continuing.

aep, 2021-01-26:
when setting up 0.18.0 from the rpi image,
there's no way to pass the setup wizard beyond "Configure the connectivity check"

when clicking next, a dialog pops up saying "You haven't yet decided on whether to enable or disable the online connectivity check. You need to either enable or disable it before continuing." 

however, the mentioned option does not appear on the current page, and you cannot change pages.
[image]
[image]

cp2004, 2021-01-26:
Can you scroll on the dialog? The buttons should be at the bottom, below the test button but it looks as if there is not enough space for them to be visible. Should be there when you scroll down.

foosel, 2021-01-26:
I don't know what you are running there, but it certainly isn't stock OctoPrint as shipped on OctoPi 0.18.0, because that looks like this at this step:

[image]

Note how half of the steps are missing in your version and the HTML looks completely different. So: What exactly did you download when and where, flashed how and did in between?

aep, 2021-01-26:
> can you scroll on the dialog?

aha! thanks. that wasn't obvious to me.

> Note how half of the steps are missing in your version and the HTML looks completely different.

Well, if you do believe that safari is "wrong", i think its fairly easy to block it and tell the user to use the "correct" browser.

> So: What exactly did you download when and where, flashed how and did in between?

i'm not sure why you'd think i'm lying but i guess the best resolution is to just close this.

foosel, 2021-01-26:
I didn't say you were lying, I didn't blame your browser, so there really is no need for this kind of reaction.

All I know is you flashed some 0.18 image and ran into a problem. I did just flash 0.18.0 as linked from the download page myself to check and could not reproduce this. So something between what you did and what I did was different and resulted in this outcome. In order to figure out what and to potentially help you, I need the questions I asked you answered. If you don't want to answer them, ok, your choice, but then I can't help you. They certainly were not meant as an offense but an attempt to figure out what led up to this result.

aep, 2021-01-26:
ok, feel free to reopen, if you agree this is confusing UX.

apple wants you to use other indicators for scrollable areas, 
but a common hack i believe is to just force them visible.

https://codepen.io/ppolyzos/pen/nDbaq

foosel, 2021-01-26:
So it in fact was a case of "scrollable but not clear that it is scrollable" and you were now able to complete the wizard?

aep, 2021-01-26:
yep exactly. works fine, thanks :)
its just confusing if you have never seen the dialog, and there's no indication that there's more options further down.
On the 2020 macbook the last button also happens to align perfectly so it look like this is intended to be the end.

foosel, 2021-01-26:
Reopening as an improvement request then.

I wasn't aware that Apple apparently decided that visible scrollbars are optional. Questionable UX choice IMHO, precisely due to issues like this one, but whatever. Will see I can implement a workaround for that in OctoPrint 1.6.0.

foosel, 2021-01-26:
The above commit enforces scrollbar visibility if Safari is detected as a browser. Will be in 1.6.0

commit 4179a6b3 (2021-01-26): 🚸 Ensure scrollbars are always visible on Safari

Closes #3987
closed 2021-04-27 by foosel