OctoPrint/OctoPrint #3988: Folder path 'test' does not validate correctly

bakgaard, 2021-01-28:
#### What were you doing?
I am working on the Docker image of octoprint/octoprint:1.5.3, though I think this affects the normal setup as well;

1. In the GUI, under ``Settings`` -> ``Folders``, 
2. I can type *anything* and still get a valid path (see image in later section)

#### What did you expect to happen?

I expect paths which do not exist, to be validated as "did not find the path".

#### What happened instead?

Path was shown as valid

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

OctoPrint 1.5.3, Python 3.8.7

#### Operating System running OctoPrint

Docker, octoprint/octoprint:1.5.3

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

Chrome, 88.0.4324.104

#### Link to octoprint.log

[Gist](https://gist.github.com/bakgaard/7ab7fec2f80e44c6ec9cfd0cca36b335).

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

Request:
```json
{"check_type":"dir","check_access":"w","allow_create_dir":true,"check_writable_dir":true,"path":"/octdfghjmk,","command":"path"}
```

... ``Allow_create_dir="true"`` ?
This might be the cause.

Is this wanted functionality, or can it be disabled?

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

cp2004, 2021-01-28:
I think they are valid because OctoPrint will create any missing folders - and that makes sense to me, otherwise the user would have to create each sub folder themselves, which isn't all that user friendly at all.

The test is instead validating that things like permissions, write access are OK. So you couldn't set the path to a root owned folder, since it would not be writable, or something like that. Likewise a folder that is owned by a different user (say on a NAS) or something like that.

bakgaard, 2021-01-28:
Perhaps we could add a "create folder if not found"-checkbox, default true?

cp2004, 2021-01-28:
Could you clarify why you would want to turn of that behaviour, and instead create the folders yourself?

foosel, 2021-01-28:
Just to chime in to say that @cp2004 is correct, this is intended behaviour. We don't want to force the user to create folder themselves if we can do it for them just fine. However, I'd be open to a change to properly reflect that a path can in theory be created but doesn't yet exist in the UI.

Idea would be that "exists" would map to "Path found" and "doesn't exist but can be created" gets mapped to a new "Path can be created", both green.

I do *not* want to change the behaviour that OctoPrint will create paths that don't exist yet if it can. That's just going to cause grief to a lot of people out there.

bakgaard, 2021-01-28:
Grief and sorrow was not intended.

I was hoping, that the check would verify, if I had already place a folder, so I would know the path was actually correct. Since it flags "exist" either way, I thought this was a bug.

With more digging, I found that my actual target was not possible anyway; I was hoping to place a folder outside my Docker container, and let the container know that this was the folder to use for `gcode` - but I've seen multiple times, that this is a "no touch" folder, and I should use the Watched folder instead.
But with the Watched folder, files get moved to the Uploads folder, and this defies my intentions. 

I believe this is no longer a bug, just not what I was looking for.

closed 2021-01-28 by bakgaard

foosel, 2021-01-29:
Just to clarify, you can mount the uploads folder from external to docker, you just should not put stuff in it yourself. OctoPrint will do its best to do its housekeeping even if stuff has been added outside of its control, but it might fail at that and that could cause issues in the process. E.g. imagine you replace the file that's currently being printed. Hence the general discouragement to fiddle around with the uploads folder, and to use the watched folder to add stuff through copying. I'm not entirely sure what your intentions here are, but if it's persistent storage AND watched folder functionality - mount both?

bakgaard, 2021-01-29:
I might try that anyway.

Thank you