OctoPrint/OctoPrint #3989: grep: Invalid content of \{\}

Stretch96, 2021-01-28:
#### What were you doing?

Restarted OctoPrint after upgrading to 1.5.3

#### What happened?

Noticed an odd line in the logs:

```
Jan 28 21:44:46 raspberrypi octoprint[1111]: grep: Invalid content of \{\}
```

OctoPrint still runs fine, appears to be caught as just a warning, just raising as an issue 👍

cp2004, 2021-01-28:
Off the top of my head, I don't know of anywhere that is using grep, I have a vague recollection of gcode analysis using it, but given how large OctoPrint is we are really going to need more info to track that down - please look at it our way, where would you start looking in the codebase with a single line from the log? The [contribution guidelines](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report) have lots of info, so you know what we are looking for 🙂/

Stretch96, 2021-01-29:
Indeed, I thought I'd raise it as an issue for now

I'll try dig into why it's happening, just in case it becomes a bigger issue in future

Stretch96, 2021-01-29:
I've manage to reproduce the issue on a blank Pi:

```
$ git clone git@github.com:foosel/OctoPrint.git
$ cd OctoPrint
$ pip install .

## /etc/systemd/system/octoprint.service
[Unit]
Description=Octoprint Server
After=network.target

[Service]
User=pi
Environment=PORT=5000
Restart=on-failure
ExecStart=/home/pi/.local/bin/octoprint serve --port=${PORT}

[Install]
WantedBy=multi-user.target

$ sudo systemctl daemon-reload
$ service octoprint start
```

Upload a file via Octoprint UI

```
sudo service octoprint restart
```

The log appears in syslog again:

```
Jan 29 00:48:07 homepi-04 octoprint[2852]: 2021-01-29 00:48:07,552 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /usr/
bin/python3 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/u
ploads/Lego_bricks.gcode
Jan 29 00:48:07 homepi-04 octoprint[2852]: grep: Invalid content of \{\}
```

I wonder if it is the gcode file that Octoprint doesn't like ...
I can provide it if needed

Stretch96, 2021-01-29:
This is the only mention of `grep` in the codebase:

```
## ./versioneer.py
def get_keywords():
    """Get the keywords needed to look up the version information."""
    # these strings will be replaced by git during git-archive.
    # setup.py/versioneer.py will grep for the variable names, so they must
    # each be defined on a line of their own. _version.py will just call
    # get_keywords().
    git_refnames = "%(DOLLAR)sFormat:%%d%(DOLLAR)s"
    git_full = "%(DOLLAR)sFormat:%%H%(DOLLAR)s"
    keywords = {"refnames": git_refnames, "full": git_full}
    return keywords
```

May be relevant 🤷

foosel, 2021-01-29:
It's probably rather the pulled in OctoPrint-FileCheck plugin, if it happens on a file upload. 

https://github.com/OctoPrint/OctoPrint-FileCheck/blob/master/octoprint_file_check/__init__.py#L76

I see absolutely no way how that can end up with an empty pattern though 🤔

So please, let's do the needful here. I need octoprint.log, a file with which to reproduce and also a test in safe mode to rule out this is a third party plugin and not filecheck after all. In short, what the template asked you to provide ;)

Stretch96, 2021-02-03:
Ah yep, apologies @foosel - I'll recreate the issue with the logs as I should have 😬 

I did just try reproduce it on the same Pi, but it's no longer producing the warning 🤔 
(Maybe that I've tainted it with other plugins or packages ... even with `--safe`)

I'll dig further and provide what is needed to hopefully fix the issue 👍

Stretch96, 2021-02-03:
Hi @foosel, I've recreated the issue, and now have logs available 👍 

Flashed a RaspberryPi, and installed Octoprint

The 'grep' error still only occurs in the syslog weirdly ...

Full 'history' log: https://octoprint-issue-3989-logs.s3.eu-west-2.amazonaws.com/logshistory.log
octoprint.log: https://octoprint-issue-3989-logs.s3.eu-west-2.amazonaws.com/logsoctoprint.log
syslog: https://octoprint-issue-3989-logs.s3.eu-west-2.amazonaws.com/logsyslog.log

foosel, 2021-02-23:
The good news is I have been able to reproduce it now. The info that it *only* shows up in syslog was crucial here. The bad news is that I still have no idea at all why this is happening. However, I'm moving this over to the FileCheck plugin since that is most certainly where this is happening.