OctoPrint/OctoPrint #3991: Consistently format/display version numers

reedy, 2021-01-30:
#### What were you doing?

Looking at "Software Updates".

#### What did you expect to happen?
Version numbers of plugins to be displayed consistently.

#### What happened instead?
Version numbers are not displayed consistently.

```
Installed: 2020.9.23
Available: 2020.09.23
```

Installed version month is not zero padded. Available is. This causes cognitive load to see that the versions are indeed the same, just formatted differently.

#### Did the same happen when running OctoPrint in safe mode?
n/a

#### Version of OctoPrint
1.8.3

#### Operating System running OctoPrint
OctoPi 0.18.0

#### Printer model & used firmware incl. version
n/a

#### Browser and version of browser, operating system running browser
n/a

#### Link to octoprint.log
n/a

#### Link to contents of terminal tab or serial.log
n/a

#### Link to contents of Javascript console in the browser
n/a

#### Screenshot(s)/video(s) showing the problem:
[image]

I have read the FAQ.

cp2004, 2021-01-30:
> Version of OctoPrint
> 1.8.3

Tell me, does it get better in the future? 😄 

I agree it's slightly off, I think it's because pip/setuptools strips leading zeros but the [tag on GitHub](https://github.com/OctoPrint/OctoPrint-FileCheck/releases/tag/2020.08.07) includes them. The least effort IMO would be to tag releases without leading zeros, rather than waste time trying to re-add them to display in the UI.

reedy, 2021-01-30:
> I agree it's slightly off, I think it's because pip/setuptools strips leading zeros but the [tag on GitHub](https://github.com/OctoPrint/OctoPrint-FileCheck/releases/tag/2020.08.07) includes them. [...]

Heh. Yeah, I'm not so fussed what the actual solution is (change the tagging in future, insert the zero in the display code, remove the zero in the display code), as long as they're consistently displayed.

I'm guessing pip/setuptools knows that the '2020.09.23' == '2020.9.23', so does some sort of normalisation somewhere... Maybe that can be called externally as part of the output, rather than just outputting straight from GitHub?

foosel, 2021-02-02:
I cannot really do any normalization on the version as fetched from the GitHub release here, as that might also contain additional things (e.g. "1.2.3 (stable)" or "1.2.3rc1 (release candidate)" or "v1.2.3"). It's the release title, not the tag (intentionally to *allow* some additional information to be shared there), freeform and depends on what the author decides to put there. But what I can do is adjust the stuff that I do control myself to be consistent here. I've just changed the titles on FirmwareCheck and FileCheck to lose the zero padding:

[image]

closed 2021-02-02 by foosel