OctoPrint/OctoPrint #3992: Internal Server Error

ingo916, 2021-01-30:
I have two raspberry pi 4's running OctoPrint, Both have the same plugins installed, identical setups. One works fine all the time but the other i get the following error when its been idle for a while. .
I can ssh to it and restarting the OctoPrint service fixes the issue. I put tape on the 5V pin per https://community.octoprint.org/t/put-tape-on-the-5v-pin-why-and-how/13574 

Could it be that OctoPrint is searching for the printer and causing some sort of timeout because there is not power to the printer? Should I remove the tape from the usb?

#### What were you doing?

Using TP-Link smartplug plugin to turn off printer after x time idle. This process works, however at some point in time i get an HTTP 500 error Internal Server Error when trying to connect to OctoPrint through a web interface.

#### What did you expect to happen?

Should be able to view the web interface

#### What happened instead?

HTTP ERROR 500

#### Did the same happen when running OctoPrint in safe mode?

I will test but i think it will have the same effect as the printer will be powered off and Octoprint will be polling the usb port serial connection.

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2021-01-30:
Please fill out the full template. Without logs and version and also the test in safe mode this can't be looked into.

ingo916, 2021-02-01:
problem did not happen in safe mode, must be a plugin. I will close this now.

closed 2021-02-01 by ingo916