OctoPrint/OctoPrint #3993: [Request] Log an error, point out max upload size, or increase max upload size for backups

cp2004, 2021-01-31:
**Is your feature request related to a problem? Please describe.**
Attempting to upload my 2.1GB backup to update from OctoPi 0.17 -> 0.18 on production, and the upload just sits there *doing absolutely nothing*.

I knew it would sit there, because I'm familiar with the codebase and know it has a max upload size of 1GB:

https://github.com/OctoPrint/OctoPrint/blob/6b516201ca2907e18803118b13522da29b53e780/src/octoprint/plugins/backup/__init__.py#L459-L461

This caught me out [over a year ago too](https://community.octoprint.org/t/backup-and-restore-not-working/14269) in one of my only actual posts on the forums, how naïve I was then. Now look where I've ended up...

**Describe the solution you'd like**
It's in the title, I think a good solution here would be:
* Error message if the file is over 1GB in size
* Point out max upload size in the backup plugin or wizard
* Increase max upload size. I don't know what would be a good value here, but I can tell you that in just over year, I accumulated 2.1GB of data.

**Additional context**
My timelapses are on Dropbox using that plugin, and I don't ever print anything twice. So I don't need this massive backup. But it sure did have me confused when it didn't do anything. 400MB one now is working beautifully. 

Nothing is logged either, by tornado. Maybe it would be in debug, but this is production and as you can guess I don't run with that enabled there.

foosel, 2021-02-02:
It's really tricky to make the right decision here. Given that a zip file can't be streamed (the inventory data structure is stored at the end), we can't check for the deflated size right at the start of the upload. If so, we could only deny uploads that exceed zip file size + extracted size. We can't however. So this turns into wildly guessing. And if a user has already created a backup with timelapse files, nuked their setup and then cannot upload that backup because it's too big, that also sucks. 

Would be very happy about some brainstorming on how to go about this. Maybe the answer is USB stick detection and putting the backup on THAT? 

In the meantime, I'm going to make the error message more sensible and point out the alternative of applying the backup from the command line as an alternative in the dialog if it exceeds the maximum size.

cp2004, 2021-02-02:
I think better error messages would be a good enough solution, whereas currently it just hangs and no upload happens. For me, it was easy enough to exclude the timelapses, either on creation, or after it is downloaded. But it sure was confusing me the first time it happened!

Once the backup is created, in this list in the UI:
[image]

Could there be a warning icon, on hover/click then notify 'this backup is too large to be uploaded again' and point out you will have to put in more work (CLI backup, remove timelapses/uploads etc.) to get it to work

foosel, 2021-02-02:
Ah, that's a great idea! We could link to an FAQ entry that explains all that. 

Gotta admit though, the USB stick idea now also is kinda stuck in my head... Will need a PR against OctoPi first though because right now a stick won't auto mount on OctoPi. Something else for the TODO list I guess ;)

foosel, 2021-02-16:
Warning's ready for 1.6.0

commit 32fd4df5 (2021-02-16): ✨ backup: add warning about max upload size

Closes #3993
closed 2021-04-27 by foosel