OctoPrint/OctoPrint #3994: M20 content is lowercased; M23 sends lowercase file names

stro, 2021-02-01:
#### What were you doing?

Octoprint 1.5.3 on RPi. Connected to SKR v1.4 board. Built a custom Marlin firmware to solve the problem with /uglywi\~1/filena\~1 in M20. 

Refreshed the list of files on SD:

`Send: M20`
`Recv: Begin file list`
`Recv: /Stands/CCR10_300_AAC_Blackout.gcode 13371377 bytes`
`Recv: End file list`
`Recv: ok`

The file in left panel is shown in lowercase:

`stands/ccr10_300_aac_blackout.gcode`
`Uploaded: -`
`Size: 12.8MB`

When trying to print it, Octoprint sends lowercased file name in M23 and naturally fails:

`Send: M23 /stands/ccr10_300_aac_blackout.gcode`
`Recv: echo:Now fresh file: /stands/ccr10_300_aac_blackout.gcode`
`Recv: open failed, File: ccr10_300_aac_blackout.gcode.`
`Recv: ok`

#### What did you expect to happen?

Keep filenames as is.

#### What happened instead?

Filenames are lowercased. 

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Version of OctoPrint

OctoPrint 1.5.3 Python 2.7.13 

#### Operating System running OctoPrint

Debian Linux on RPi3

Linux raspberrypi 4.19.66-v7+ \# 1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux

#### Printer model & used firmware incl. version

Sunlu S8; SKR v1.4 Turbo board; Marlin bugfix-2.0.x with experimental long file name support that I try to incorporate in the firmware.

foosel, 2021-02-02:
It looks like this lower casing crept in while implementing support for associating long names from `M33` with short names from `M20`, but I'm not sure right now if that was intentional or an accident and have to investigate.

However:

```
Recv: /Stands/CCR10_300_AAC_Blackout.gcode 13371377 bytes
```

That is a wrong output format in any case. The `bytes` prefix there needs to go and doesn't belong. Did you add that or did that make its way into some Marlin variant? Because that will seriously break stuff not only for OctoPrint.

It should be

```
Recv: /Stands/CCR10_300_AAC_Blackout.gcode 13371377
```

It's implicit that those are bytes. The format is

    <filename>[ <filesize in bytes>[ <longname>]]

At least OctoPrint won't have a problem with having the long name in place of the usual 8.3 filename (what's important here is that the firmware accepts whatever is listed there as input parameter to select/delete/etc commands), but there should definitely not be a `bytes` tacked on there.

foosel, 2021-02-02:
Looked a bit further and found various locations at least in Marlin's source that made it sound like `<filename>` is supposed to be case insensitive. Hence probably my lowercasing it. Of course that doesn't work if you throw the case sensitive longname into its place. I'll still revert the lower casing and see if there are any issues with that during the RC phase.

Looking again at your patch however, you really should not put the longname into position of `<filename>` after all and rather extend M20 to stick to the above format. If your long name contains whitespace (which it may, 8.3 may not), things will break.

stro, 2021-02-04:
[M20-long-names.zip](https://github.com/OctoPrint/OctoPrint/files/5922891/M20-long-names.zip)

Here's the current diff. I removed " bytes" from the code already, and omitted some changes related to the printer itself.

commit dc2b63ba (2021-02-02): 🥅 comm: treat sd file names case sensitive

Just in case some firmware variant decides to go for case sensitivity
here (shouldn't usually happen since this should always be 8.3 names,
but people tinker...) we treat the SD file names case sensitively again.

Closes #3994
closed 2021-04-27 by foosel