OctoPrint/OctoPrint #3999: tests/settings/test_settings.py::TestSettings::test_set_external_modification - AssertionError: '127.0.0.1' != '0.0.0.0'

ByReaL, 2021-02-03:
git clone ..... octoprint
pytest ./tests

1 in 10 that test will fail, looks to be intermittent
looks to me like a test bug.

________________________________________________________________________________________________ TestSettings.test_set_external_modification ________________________________________________________________________________________________

self = <settings.test_settings.TestSettings testMethod=test_set_external_modification>

    def test_set_external_modification(self):
        with self.mocked_config() as paths:
            basedir, configfile = paths
            settings = octoprint.settings.Settings()

            self.assertEqual("0.0.0.0", settings.get(["server", "host"]))

            # modify yaml file externally
            config = self._load_yaml(configfile)
            config["server"]["host"] = "127.0.0.1"
            self._dump_yaml(configfile, config)

            # set some value, should also reload file before setting new api key
            settings.set(["api", "key"], "key")

            # verify updated values
>           self.assertEqual("127.0.0.1", settings.get(["server", "host"]))
E           AssertionError: '127.0.0.1' != '0.0.0.0'
E           - 127.0.0.1
E           + 0.0.0.0

tests/settings/test_settings.py:620: AssertionError

cp2004, 2021-02-07:
Approving manually, since I don't think there is much else needed - I haven't seen this myself but was told about it on discord once.

@ByReaL Maybe you could provide information about the install? What OS? What branch/version? Anything that will make life a bit easier.

foosel, 2021-02-08:
Some additional info here would certainly help, because I've never seen this test fail in my own runs or the automatic runs that are triggered on every single push 😶

shadycuz, 2021-02-20:
@foosel 

This failed for me in my PR #3903 

Look at my very first description. 

[image]

I will see if I can reproduce and try to fix it. 

@ByReaL are you using WSL?

shadycuz, 2021-02-20:
Very intermittent 🤔 

```
(octobug) [~/repos/shadycuz/octo-bug]
shadycuz-> pytest tests/settings/test_settings.py 
================================================================================================================================================ test session starts ================================================================================================================================================
platform linux -- Python 3.9.0, pytest-4.6.10, py-1.10.0, pluggy-0.13.1
rootdir: /home/shadycuz/repos/shadycuz/octo-bug, inifile: pytest.ini
plugins: doctest-custom-1.0.0
collected 65 items                                                                                                                                                                                                                                                                                                  

tests/settings/test_settings.py .................................................................                                                                                                                                                                                                             [100%]

============================================================================================================================================= 65 passed in 3.11 seconds =============================================================================================================================================
(octobug) [~/repos/shadycuz/octo-bug]
shadycuz-> pytest tests/settings/test_settings.py 
================================================================================================================================================ test session starts ================================================================================================================================================
platform linux -- Python 3.9.0, pytest-4.6.10, py-1.10.0, pluggy-0.13.1
rootdir: /home/shadycuz/repos/shadycuz/octo-bug, inifile: pytest.ini
plugins: doctest-custom-1.0.0
collected 65 items                                                                                                                                                                                                                                                                                                  

tests/settings/test_settings.py ..................................................F..............                                                                                                                                                                                                             [100%]

===================================================================================================================================================== FAILURES ======================================================================================================================================================
____________________________________________________________________________________________________________________________________ TestSettings.test_set_external_modification ____________________________________________________________________________________________________________________________________

self = <settings.test_settings.TestSettings testMethod=test_set_external_modification>

    def test_set_external_modification(self):
        with self.mocked_config() as paths:
            basedir, configfile = paths
            settings = octoprint.settings.Settings()
    
            self.assertEqual("0.0.0.0", settings.get(["server", "host"]))
    
            # modify yaml file externally
            config = self._load_yaml(configfile)
            config["server"]["host"] = "127.0.0.1"
            self._dump_yaml(configfile, config)
    
            # set some value, should also reload file before setting new api key
            settings.set(["api", "key"], "key")
    
            # verify updated values
>           self.assertEqual("127.0.0.1", settings.get(["server", "host"]))
E           AssertionError: '127.0.0.1' != '0.0.0.0'
E           - 127.0.0.1
E           + 0.0.0.0

tests/settings/test_settings.py:620: AssertionError
======================================================================================================================================== 1 failed, 64 passed in 3.07 seconds ========================================================================================================================================
````

shadycuz, 2021-02-20:
I do not know the root cause, but I know where it's happening and I can fix it. @foosel  which branch would you like the bugfix to be based on? master? 

In the test you see:
```python
# set some value, should also reload file before setting new api key
settings.set(["api", "key"], "key")
```

but the file is not being reloaded. In the `Settings` class the `set` method has this condition on the call to load:
```python
if self._mtime is not None and self.last_modified != self._mtime:
            self.load()
```

I first removed the conditional and had it call `load()` everytime and the bug went away. I then printed out the values to see what it was doing.

```python
if self._mtime is not None and self.last_modified != self._mtime:
            self.load()

print(self._mtime)
print(self.last_modified)
# self.load()
```

```
>           self.assertEqual("127.0.0.1", settings.get(["server", "host"]))
E           AssertionError: '127.0.0.1' != '0.0.0.0'
E           - 127.0.0.1
E           + 0.0.0.0

tests/settings/test_settings.py:626: AssertionError
----------------------------------------------------------------------------------------------------------------------------------------------- Captured stdout call ------------------------------------------------------------------------------------------------------------------------------------------------
1613813683.6274326
1613813683.6274326
```

It would seem like if your filesystem is slow enough, but your processer is fast enough you can bypass the conditional in the `set` method. 

This seems like something that would never happen in the actual operation of OctoPrint but something only possible during the test case. 

This is the fix I propose unless you can find something more elegant. 

```python
def test_set_external_modification(self):
        with self.mocked_config() as paths:
            basedir, configfile = paths
            settings = octoprint.settings.Settings()
            sleep(0.1)
            ```

foosel, 2021-02-20:
That would be my suggestion as well, especially considering that the test here is supposed to test the conditional reload.

Bugfixes, even for bugs in test cases, always again `maintenance` please - thank you! 👍

foosel, 2021-02-22:
Merged. And since this is a unit test issue and not a bug in the code per se, I'm going to close this now since there's no specific release needed for effectiveness. `maintenance` and soon `devel` branches should be fine now.

commit a628549a (2021-02-22): Fixes #3999 test fails to reload config file
closed 2021-02-22 by foosel