OctoPrint/OctoPrint #4000: Version 1.5.3 Print Time Estimation

criton01, 2021-02-04:
Since upgrading to Version 1.5.3 the print time estimation is WAY off...  11.9mb file estimates at 2.5hrs
[image]
 but print time while printing is in 2 days.

jneilliii, 2021-02-04:
Notice the red dot, that means that estimation is not accurate. Recommend you look into the PrintTimeGenius plugin, as you will get far more accurate estimates than what's included by default in OctoPrint as it learns over time your specific printer and how fast it prints in real life. This isn't necessarily an issue, and because there is a red dot indicating inaccurate estimates for whatever reason (maybe the analysis didn't complete prior to starting the print), going to ho ahead and close this one. If you do think this is an actual bug, then please re-open the ticket with the full requested information in the issue template.

closed 2021-02-04 by jneilliii

criton01, 2021-02-04:
I’ll load the plugin just to check it out… it only happened AFTER upgrade to 1.5.3  I’ll get the template and submit if it’s a real problem.

Thanks for responding.

cp2004, 2021-02-04:
Please always state what version you upgraded *from* as well as where you are at now. If you upgraded from 1.4.x then yeah there's been quite a few changes to OctoPrint. But if it was 1.5.2 -> 1.5.3 then no, only a couple lines changed relating to plugin install.