OctoPrint/OctoPrint #4001: Allow newer zeroconf releases

fabaff, 2021-02-04:
The latest release of zeroconf is 0.28.8. The current version limitation causes issues if one want to install `octoprint` in the same `venv` with other tools like Home Assistant.

https://github.com/OctoPrint/OctoPrint/blob/cc876e3794553d35471772df663b485b26853484/setup.py#L82

For NixOS is required to ignore the version constraints as well.

fieldOfView, 2021-02-04:
> The current version limitation causes issues if one want to install octoprint in the same venv with other tools like Home Assistant.

Then... don't do that. Run them both in different virtualenvs; that's the whole idea of a virtualenv.

foosel, 2021-02-04:
I think we need to talk a bit about virtual environments and OctoPrint's supported setups.

OctoPrint's setup guide says to create a dedicated virtual environment for it. That is not only the recommended but actually  the only supported way of setting it up, precisely because I cannot control the possible dependency versions available in a myriad of OS distributions or required by other software.

You should not be running OctoPrint in the same venv as HA. And you also should not be running OctoPrint again distribution dependencies. You should be using a virtual environment.

If you want to set it up at OS level, you may of course do that, but if any dependency conflicts - like this one and also #4002 - result from that, then that's too bad but not something I can justify spending any time be on analysing and/or fixing. I'm happy to accommodate even more unique environments where I can, but there are limits, and supporting stuff that explicitly goes again the officially supported setup environment is beyond that.

I update dependencies as far as I can justify before every release, while taking current environment limitations into account. I will do so as well for watchdog and zeroconf, as needed, and where it doesn't cause diverging code bases.

closed 2021-02-04 by foosel

fabaff, 2021-02-05:
> You should not be running OctoPrint in the same venv as HA.

Sure, but users do what they like and often do not know better unfortunately. 

> And you also should not be running OctoPrint again distribution dependencies.

Well, most application developers say that :wink:. From an end-user point of view using the globally available distribution packages is preferred in multiple ways.

One can simply `dnf -y install octoprint` or `nix-env -iA nixpkgs.octoprint` and is ready to go. No dealing with Python, compilers, containers, additional dependencies and development libs. No need to take care manually about updates either.

There are people out there who are running OctoPrint on NixOS that way since 1.2.9 or approx. 5 years. 

> that's too bad but not something I can justify spending any time on analysing and/or fixing.

I'm not asking to fix or debug anything. The debugging was already done and the result is that using out-dated requirements may diminish the user experience as the issue has to be worked around.

> I'm happy to accommodate even more unique environments where I can, but there are limits, and supporting stuff that explicitly goes against the officially supported setup environment is beyond that.

Again, you don't have to accommodate anything. Sorry, I don't get it. Why should supporting up-to-date requirements go "explicitly against the officially supported setup"?

foosel, 2021-02-05:
You explicitly phrased your request "Allow newer zeroconf releases", "The current version limitation causes issues if one want to install octoprint in the same venv with other tools like Home Assistant." and "For NixOS is required to ignore the version constraints as well.".

This together with your other ticket - to me - sounded like "make your requirements wider in general so they work in arbitrary unsupported setups" and not just "take a look if whatever is stable at the next release point is time might qualify". The latter is something that's done anyhow whenever I can (and involves hunting down and studying changelogs, and reading through commitlogs where those don't exist), the former is something that will increase development overhead (sadly, a ton of those dependencies do not stick to semantic versioning). 

As I said, I do regularly update the min and max version requirements leading up to releases. In the case of zeroconf, I am very conservative about this however since it is incredibly expensive to test that everything still works after a library update (I recently spent the better part of a week debugging weird mdns issues as a consequence that had me write custom tooling to even begin to grasp what was going on), we are stuck with an older version under python 2 that has to be kept API compatible with the py3 build manually and that means that here I frankly stay very conservative with updates, as long as I'm not aware of any security issues or crippling bugs. Not having a readily available changelog that I can consult to decide whether updating the dep causes a torrent of bugs also doesn't help here (if you are aware of a changelog that isn't the git commit log, I'd be happy about this resource). tldr: with this dependency in particular, it is too damn risky to "just update it" if the only reason to do so is "I want to be able to install that in an unsupported setup".

> Well, most application developers say that

Yes, because we only scale so much. We cannot support every possible setup that people come up with under the sun. There's a recommendation. Anything apart from that is a case of "there be dragons". OctoPrint is already quite broadly setup here considering that it's OS agnostic, that already can be quite the challenge.