OctoPrint/OctoPrint #4002: Allow newer watchdog releases

fabaff, 2021-02-04:
The latest watchdog release is [1.0.2](https://pypi.org/project/watchdog/). They change their versioning schema.

https://github.com/OctoPrint/OctoPrint/blob/cc876e3794553d35471772df663b485b26853484/setup.py#L49

With 1.0.0 support for Python 2.7, 3.4 and 3.5 was dropped. Python 3.6 is still [supported](https://github.com/gorakhargosh/watchdog/blob/master/setup.py#L152).

cp2004, 2021-02-04:
OctoPrint will still support Python 2.7 until 2.0 is released, which is currently in progress on the devel branch. We can't drop Python 2.7 at the moment, since the upgrade of all existing installs is only just underway, at 20% now, increasing every day.

closed 2021-02-04 by cp2004

cp2004, 2021-02-04:
To add - Watchdog 1.0.0 is pinned on the devel branch, since that is Python 3.6+ and therefore compatible. It is pinned to a specific version because they created a breaking change on a point release, which meant OctoPrint refused to install on Python 3 systems. This meant a rushed hotfix release had to be created, and to avoid any such issues in the future it is currently pinned exactly.
Of course, this could be changed in the future but there is no way for us to tell they won't do the same thing again now.

fabaff, 2021-02-04:
I don't get it...you have `INSTALL_REQUIRES_PYTHON2` and `INSTALL_REQUIRES_PYTHON3`, thus for Python 2 you could stick to `watchdog==0.10.4` and for Python 3 support later releases.

cp2004, 2021-02-04:
It's not ideal to have different versions of dependencies for different Python versions, it was done for feedparser because there's no release that will support both Python 2.7 and 3.9, unsure about zeroconf. 
Could you provide some more details about why you need watchdog to be of a higher version? Even if it was pinned to 1.0.2, what benefits does this bring?

fabaff, 2021-02-04:
Because the NixOS set package has 1.0.2 now. To make the `octoprint` build pass, the version constrain of `watchdog` and others (e.g., `emoji` or [`zeroconf`](https://github.com/OctoPrint/OctoPrint/issues/4001)) is ignored. This is not ideal as it's a deviation from what `octoprint` wants.

If `octoprint` would allow `watchdog<1.1.0` then builds would pass and the NixOS package could stay closer to the upstream requirements. I know this is a risk as it happen that dependencies introduce breaking changes with minor releases.

cp2004, 2021-02-04:
Request for comment from @foosel here, I personally would not be too keen on it for a while but since it is not my project, I don't make the decisions 🙂. I do see this being a reoccurring issue each time there is an update to NixOS packages, though I don't know much about watchdog or NixOS's development to know how often it is updated.

foosel, 2021-02-04:
See here: 

https://github.com/OctoPrint/OctoPrint/issues/4001#issuecomment-773529770