OctoPrint/OctoPrint #4003: [Request] Temperature form auto-select or empty

davinderc, 2021-02-04:
**Is your feature request related to a problem? Please describe.**
Whenever I select one of the target temperature input forms in the Temperature tab, I am met with a form that already has a zero in it, but it is not selected. I am accustomed to the pre-filled text in a form being auto-selected when I place focus in the form, so I type in my temperature of 60 C for the bed and end up with it going to 600 C. It's a minor inconvenience, as I always see that and correct it (plus the maximum temperature that Octoprint will set my printer bed to is 110 C). The same happens for the extruder nozzles, although strangely for those I seem to set the temperature more easily.

**Describe the solution you'd like**
I would like for the forms to either be empty and just contain the cursor so that anything I type in won't have an extra zero at the end. Another option would be to simply select the text inside the form when I click on it, so that anything I type will overwrite it. Of course, this may simply be a matter of preference, so maybe there could be a setting for these types of forms?

**Describe alternatives you've considered**
The other option I can think of is to simply live with it and learn to expect that extra zero.

**Additional context**
I just spent some time clicking on the form, and it seems the behavior is not predictable. Sometimes when I click on it, only part of the text gets selected. Sometimes the cursor appears next to the text that is already pre-filled there. And other times the entire text gets selected when I put focus on the form, which is what I would like to happen all the time. Maybe this is actually a bug?

Behaviours I've seen are shown in the screenshots below:

Behaviours I don't prefer:
[image]
[image]

Behaviour I would like to see happen every time I click in the form:
[image]

nivekmai, 2021-02-05:
It might be useful to know what browser you're using to view the site.

Taking a quick glance at the code, it seems it should be selecting the whole text on focus:

https://github.com/OctoPrint/OctoPrint/blob/4bc2435412f1f7a99042454822a940164cd9c5d0/src/octoprint/static/js/app/viewmodels/temperature.js#L948

https://developer.mozilla.org/en-US/docs/Web/API/HTMLInputElement/select

As an anecdotal data point, I can not reproduce this issue on Chrome 88.0.4324.104. Every time I focus an input in the temperature control, the entire text content is selected.

foosel, 2021-02-05:
I cannot reproduce this behaviour in Firefox 85.0 either, it behaves exactly as requested here.

So yep, we need more information here. As things are, this is already implemented.

davinderc, 2021-02-05:
Is it possible this works differently depending on whether one is on a Mac or PC? I'm running Firefox 85.0.1 (64-bit) on Catalina 10.15.7.

foosel, 2021-02-08:
I just tested this against Firefox 85 on Catalina via browserstack, and I can't reproduce it there either.

davinderc, 2021-03-09:
My apologies Gina, I kind of forgot that I posted this 😅 
I've just popped up my printer and Octoprint. I am still seeing an issue, but I've tried it out on Chrome as well (also on Mac) and it seems the same.

However, I might try it out on my Windows machine tonight, as it seems that depending on how I click I see slightly different behavior. It may even be the Magic Mouse's debounce mechanism. So I've tried clicking "slowly", so to speak (or gently?), and I get a selected "0" in the input box (before it showed "off"). When I click normally/quickly, it seems that for a split second the 0 is selected and then it gets unselected, which seems to be that two clicks are happening. 

What's strange is that I've never observed this behavior anywhere else. Even in this Github text box I need to click twice on a word for it to be selected and three times to select the entire line.

I hope you understood what I meant. I'll also see if I can get a short screen recording later to show you what I am seeing.

foosel, 2021-03-09:
Maybe there's a misunderstanding here. OctoPrint is set up so that a *single* click in this input will select everything in it: 

[image]

That's also what I understood you were requesting, and then understood was not already working for you (because otherwise why would you be requesting that?).

Can you clarify please?

davinderc, 2021-03-09:
Yes, I am only performing a single click on the input boxes. I've created a GIF below (sorry for the poor recording/editing). The first attempt gives the desired result, the second does not. I then recorded further attempts to show that it is not related to a specific box.

I'm sorry I've been unable to be clearer, I'm not sure where I could get more information on exactly what is happening. Are there logs somewhere or some kind of debugging interface where I can see what happens when I click? I was thinking the developer tools on Firefox, but I'm not sure where I can find the info I need.
[image]

jneilliii, 2021-03-09:
Does it have to be set (ie push the button to activate that temp) and then it works as expected?

davinderc, 2021-03-09:
@jneilliii I'm not sure what you mean. In the GIF above I move the cursor over the box and click once. As far as I can tell, I can't precisely pinpoint what is different (at first I thought it might have something to do with my mouse, but in other apps it seems to behave normally)

jneilliii, 2021-03-09:
That's not the issue though. It works as expected when I just tested. I click in the input for temp and it always selects all the text.

davinderc, 2021-03-11:
I could just close this, as it doesn't seem worth it to spend a lot of time/effort if I can't find any more info to debug this. Do you agree @foosel @jneilliii ?

jneilliii, 2021-03-11:
yeah, if we can't reproduce the issue it's specific to your environment. Could be something like a security extension in the browser that prevents certain actions or something like that. You could try in incognito mode to see if that is the case.

One thing I haven't seen from you though is did you check the browser's developer console (F12) for any errors on initial page load or during the clicking process. Also, did you try running OctoPrint in safe mode?

closed 2021-05-03 by foosel